lgsvl/simulator #615: Simulator built from source crashing (Ubuntu 16.04)

DaPineappleDude, 2020-01-30:
Following the [instructions](https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md) on building the simulator from source. The build proceeds without any error and Unity shows that build completes but I am running into these problems:

-  When I launch the simulator and selecting the graphics settings the unity window crashes after showing Unity logo and turning black. As a reference, I can run the pre-built simulator release with no problems.
[image]

- The `Web` folder in the final build is missing even though I do `npm install` and `npm run pack`
[image]

The reason I am rebuilding the simulator is because I wanted to use a G29 steering wheel and had to modify the Controller Inteactions related stuff.

EricBoiseLGSVL, 2020-01-31:
@DaPineappleDude Do you have the Editor logs to see if there are any errors in the build process?

Did you set the OS to Linux?

DaPineappleDude, 2020-01-31:
@EricBoiseLGSVL I didn't see any errors during the build and it showed `Player build successful` on Unity Editor. I will check if I have the logs and get back.

Yes, I did set the OS to Linux.

Also, after some digging I noticed a folder `~/simulator/` with `AssetBundles` and `Web` as sub folders inside my source code directory. I found that odd.

EricBoiseLGSVL, 2020-01-31:
Ok, if no errors on build let's look at player.log to see what is happening.  AssetBundles and Web sub folders is expected.  AssetBundles folder holds built asset, e.g. environments, vehicles, etc. and Web folder holds WebUI assets for sign in and simulation settings.

Also, please see what nvidia drivers version you are on and did you build an environment bundle and vehicle bundle?  If there are no environments or vehicles to load it might be crashing.  The best way to quickly add the latest is to download latest binary release, run it and it will open WebUI that will download all public released assets.  If that fails then delete the LG Silicon Valley Lab folder in AppData folder in user files and run binary again.

closed 2020-04-08 by EricBoiseLGSVL