lgsvl/simulator #617: UNITYTLS_X509VERIFY_FLAG_NOT_TRUSTED when starting simulator

varesa, 2020-01-30:
I've downloaded and tried the binary releases of lgsvlsimulator-linux64-2019.11 and lgsvlsimulator-linux64-2019.12 on Fedora 31 and both go into a redirect loop in the login/authentication phase.

The window with "Open Browser..." appears and the button goes to https://account.lgsvlsimulator.com/?returnUrl=http%3A%2F%2Flocalhost%3A8080%2F

The browser is redirected to `http://localhost:8080/?token=<redacted>` and soon does a PUT request to `http://localhost:8080/users/<redacted>`.

This request fails with status 500 and a body of `{"error":"Failed to add/update user by token: <redacted>, An error occurred while sending the request"}`

Later the client tries to `GET /users` which fails 401 with `{"error":"User is not authorized: Exception of type 'Nancy.ErrorHandling.RouteExecutionEarlyExitException' was thrown.","cloudUrl":"https://account.lgsvlsimulator.com"}`. The page redirects to https://account.lgsvlsimulator.com again and loops forever.

The following errors show up in `~/.config/unity3d/Editor/Player.log`:
```
Updating user with token
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

TlsException: Handshake failed - error code: UNITYTLS_INTERNAL_ERROR, verify result: UNITYTLS_X509VERIFY_FLAG_NOT_TRUSTED
  at Mono.Unity.Debug.CheckAndThrow (Mono.Unity.UnityTls+unitytls_errorstate errorState, Mono.Unity.UnityTls+unitytls_x509verify_result verifyResult, System.String context, Mono.Security.Interface.AlertDescription defaultAlert) [0x00036] in <0079a30f96a047348857e1cecc6c638a>:0 
  at Mono.Unity.UnityTlsContext.ProcessHandshake () [0x00082] in <0079a30f96a047348857e1cecc6c638a>:0 
  at Mono.Net.Security.MobileAuthenticatedStream.ProcessHandshake (Mono.Net.Security.AsyncOperationStatus status) [0x0003e] in <0079a30f96a047348857e1cecc6c638a>:0 
  at (wrapper remoting-invoke-with-check) Mono.Net.Security.MobileAuthenticatedStream.ProcessHandshake(Mono.Net.Security.AsyncOperationStatus)
  at Mono.Net.Security.AsyncHandshakeRequest.Run (Mono.Net.Security.AsyncOperationStatus status) [0x00006] in <0079a30f96a047348857e1cecc6c638a>:0 
  at Mono.Net.Security.AsyncProtocolRequest+<ProcessOperation>d__24.MoveNext () [0x000ff] in <0079a30f96a047348857e1cecc6c638a>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.ConfiguredTaskAwaitable+ConfiguredTaskAwaiter.GetResult () [0x00000] in <7d97106330684add86d080ecf65bfe69>:0 
  at Mono.Net.Security.AsyncProtocolRequest+<StartOperation>d__23.MoveNext () [0x0008b] in <0079a30f96a047348857e1cecc6c638a>:0 
Rethrow as AuthenticationException: A call to SSPI failed, see inner exception.
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <7d97106330684add86d080ecf65bfe69>:0 
  at Mono.Net.Security.MobileAuthenticatedStream+<ProcessAuthentication>d__47.MoveNext () [0x00254] in <0079a30f96a047348857e1cecc6c638a>:0 
Rethrow as AggregateException: One or more errors occurred.
  at System.Threading.Tasks.Task.ThrowIfExceptional (System.Boolean includeTaskCanceledExceptions) [0x00011] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Threading.Tasks.Task.Wait (System.Int32 millisecondsTimeout, System.Threading.CancellationToken cancellationToken) [0x00043] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Threading.Tasks.Task.Wait () [0x00000] in <7d97106330684add86d080ecf65bfe69>:0 
  at Mono.Net.Security.MobileAuthenticatedStream.AuthenticateAsClient (System.String targetHost, System.Security.Cryptography.X509Certificates.X509CertificateCollection clientCertificates, System.Security.Authentication.SslProtocols enabledSslProtocols, System.Boolean checkCertificateRevocation) [0x0000d] in <0079a30f96a047348857e1cecc6c638a>:0 
  at Mono.Net.Security.MonoTlsStream.CreateStream (System.Byte[] buffer) [0x0007b] in <0079a30f96a047348857e1cecc6c638a>:0 
  at System.Net.WebConnection.CreateStream (System.Net.HttpWebRequest request) [0x00073] in <0079a30f96a047348857e1cecc6c638a>:0 
Rethrow as WebException: Error: TrustFailure (One or more errors occurred.)
  at System.Net.HttpWebRequest.EndGetRequestStream (System.IAsyncResult asyncResult) [0x0003a] in <0079a30f96a047348857e1cecc6c638a>:0 
  at System.Threading.Tasks.TaskFactory`1[TResult].FromAsyncCoreLogic (System.IAsyncResult iar, System.Func`2[T,TResult] endFunction, System.Action`1[T] endAction, System.Threading.Tasks.Task`1[TResult] promise, System.Boolean requiresSynchronization) [0x0000f] in <7d97106330684add86d080ecf65bfe69>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.ConfiguredTaskAwaitable`1+ConfiguredTaskAwaiter[TResult].GetResult () [0x00000] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Net.Http.HttpClientHandler+<SendAsync>d__64.MoveNext () [0x002e7] in <09b222309a944a9a9cd1b50a05dde50d>:0 
Rethrow as HttpRequestException: An error occurred while sending the request
  at System.Net.Http.HttpClientHandler+<SendAsync>d__64.MoveNext () [0x00478] in <09b222309a944a9a9cd1b50a05dde50d>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.ConfiguredTaskAwaitable`1+ConfiguredTaskAwaiter[TResult].GetResult () [0x00000] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Net.Http.HttpClient+<SendAsyncWorker>d__47.MoveNext () [0x000ca] in <09b222309a944a9a9cd1b50a05dde50d>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <7d97106330684add86d080ecf65bfe69>:0 
  at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <7d97106330684add86d080ecf65bfe69>:0 
  at Simulator.Web.Modules.UsersModule+<>c__DisplayClass0_0+<<-ctor>b__1>d.MoveNext () [0x0017a] in <094d2de7324140cea54b2e0aa4d32af5>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
[... 35 lines cut ...]
System.Threading.Tasks.UnwrapPromise`1:TrySetFromTask(Task, Boolean)
System.Threading.Tasks.UnwrapPromise`1:InvokeCore(Task)
[... cut ...]
System.Threading.QueueUserWorkItemCallback:WaitCallback_Context(Object)
System.Threading.ExecutionContext:RunInternal(ExecutionContext, ContextCallback, Object, Boolean)
System.Threading.ExecutionContext:Run(ExecutionContext, ContextCallback, Object, Boolean)
System.Threading.QueueUserWorkItemCallback:System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
System.Threading.ThreadPoolWorkQueue:Dispatch()
System.Threading._ThreadPoolWaitCallback:PerformWaitCallback()
 
(Filename: <0079a30f96a047348857e1cecc6c638a> Line: 0)
```

EricBoiseLGSVL, 2020-01-31:
@varesa We haven't seen this yet. Thanks for letting us know and posting the Player.log.
@mark-gerow-lge Can you see what might be the issue?

zelenkovsky, 2020-01-31:
@varesa Is your date and time correct on your machine?

After successful sign-in inthe cloud it tries to update your local simulator with token provided by the cloud. But that operation fails giving:

```
{"error":"Failed to add/update user by token: <redacted>, An error occurred while sending the request"}
```

Could you please attach a complete Player.log file. We are planning to get rid of local WebUI in a favor of proper cloud web interface in matter of months, so that whole thing will be gone soon.

varesa, 2020-01-31:
@zelenkovsky, date and time is correct (synchronized via NTP)

[Full Player.log](https://github.com/lgsvl/simulator/files/4140765/Player.log.txt)

I also tried with both the "DEFAULT" (which seems to limit things like allowed old TLS versions/ciphers) and "LEGACY" system crypto-policies with the same effect.

etendue, 2020-02-13:
I have the same error.
When I work on machine without https_proxy it works fine.
When on machine with https_proxy the error happens.

etendue, 2020-02-13:
I don't think it is related to proxy directly. Because when I disable proxy and use a moble connection, the problem still exists. I doubt it deals with some browser security or maybe again proxy security.

varesa, 2020-02-13:
I do not have a proxy configured either. 

I tested with the 2020.1 release from 2 days ago, still the same issue

etendue, 2020-02-17:
Finally I know where is the problem, it is not in WebUI, but in Web modules (cs code) using Nancy library. The exception comes from follow code 
`
               try
                {
                    var client = new HttpClient();
                    client.DefaultRequestHeaders.Add("Accept", "application/json");

                    var postData = new [] { new KeyValuePair<string, string>("token", x.token) };
                    var formContent = new FormUrlEncodedContent(postData);

                    var response = await client.PutAsync(Config.CloudUrl + "/users/token", formContent);
                    var content = await response.Content.ReadAsStringAsync();

                    var json = JSONNode.Parse(content);
                    UserModel userModel = new UserModel()
                    {
                        Username = json["username"].Value,
                        SecretKey = json["secretKey"].Value,
                        Settings = json["settings"].Value
                    };
                    userService.AddOrUpdate(userModel);
                    SIM.InitUser(userModel.Username);

                    var guid = Guid.NewGuid();
                    UserMapper.RegisterUserSession(guid, userModel.Username);
                    return this.LoginWithoutRedirect(guid, DateTime.UtcNow.AddSeconds(Config.sessionTimeout));
                }
               catch (Exception ex)
                {
                    Debug.LogException(ex);
                    return Response.AsJson(new { error = $"Failed to add/update user by token: {x.token}, {ex.Message}" }, Nancy.HttpStatusCode.InternalServerError);
                }
`
I am not familiar with Nancy library  and add support for SSLProxy if it was the problem

etendue, 2020-02-20:
I finally got it work.  
I just add the proxy information to httpClient(). It is not a elegant solution, but it works.

closed 2020-10-06 by EricBoiseLGSVL