lgsvl/simulator #618: Saving sensor data from "SensorNameSensor.cs"

SeymurSm, 2020-01-31:
Hi Guys,

I am trying to save sensor data without using PythonApi. There is specific Save method on every sensor script. For example for color camera it is 
 ```
public bool Save(string path, int quality, int compression)
{
.......
}
```
Unity crashes when there is reference to this method with some correct parameters. Is there something special should be done to use these Save methods?

EricBoiseLGSVL, 2020-01-31:
@SeymurSm @daviduhm Do you know why this might be happening?

SeymurSm, 2020-02-03:
@EricBoiseLGSVL sorry for late reponse. I thought it was because of the function running on UI thread when we call it directly on key click or something like that. I call the method on the background thread it is also crash Unity. There is nothing logical comes to my mind.

EricBoiseLGSVL, 2020-02-03:
@SeymurSm Do you have a player.log that we can look at?
What are you using to save the sensor data then?  If you are not using the API, you shouldn't call this method, esp on the main thread (UI events).  If you can give us some more info on what you are trying to accomplish, we can be more helpful.

closed 2020-10-20 by EricBoiseLGSVL