lgsvl/simulator #619: Apollo 5.0 Planning module crashes

danyz91, 2020-01-31:
Hi all,
I am trying to run Apollo 5.0 stack alongside LGSVL simulator.
I have followed the guide and everything works fine when simulator and stack are on the same machine. 
But I experienced some delays in Apollo execution and I decided to run the simulator on a different machine. 
With this configuration the planning module crashes when routing request succeds and it is expected to produce the trajectory.
My setup is :

- LGSVL simulator version 2019.10 running on Windows 10
- Apollo 5.0 stack on Ubuntu 16.04

Does anyone have any suggestions about this problem?
Thanks all

EricBoiseLGSVL, 2020-01-31:
@danyz91 @hadiTab Do you know why this might be happening?

shalinmehtalgsvl, 2020-01-31:
Hi @danyz91 ,

We have experienced similar issues when the computers do not have synced time. The Simulator cannot be ahead of the computer Apollo is on (Apollo uses system time). I just change the Ubuntu time to be 1-2 minutes ahead of the Windows machine. 

If this is not the issue, try the following:

On the Ubuntu machine, instead of enabling planning from Dreamview, can you try starting in the terminal. In a terminal window, enter the apollo docker then run `planning.sh start_fe`. That should give you the error that causes planning to crash.

closed 2020-04-08 by EricBoiseLGSVL