lgsvl/simulator #620: cannot open lgsvl

wangyixu14, 2020-02-03:
I am using google cloud compute engine on Ubuntu 18.04  with GPU Nvidia information shown as follows:
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2     |
+-----------------------------------------------------------------------------+
When I click ok, no windows show up for open the brower.
The ~/.config/unity3d/Editor/Player.log shows as:

Vulkan detection: 0
No supported renderers found, exiting.
 
But I do have used the  'sudo apt-get install libvulkan1 ' and successfully executed. 

What's the problem here?

EricBoiseLGSVL, 2020-02-04:
Cloud is not technically supported at the moment but @herb-kuta-lge might be able help.

wangyixu14, 2020-02-04:
> Cloud is not technically supported at the moment but @herb-kuta-lge might be able help.

herb-kuta-lge, 2020-02-06:
Hi. I've found that an X server must be running on the same machine as the simulator and the `DISPLAY` environment variable set to the display it's serving in order for Unity to find a Vulkan renderer.

QLV5645-QLJ, 2020-02-06:
Hi, thanks for your replying. But it still has the same problem.

I am able to get this window on my local computer. But when I click Ok, nothing happen, How do you run that?
[image]

herb-kuta-lge, 2020-02-06:
Is Simulator running on your local computer? What is the value of `DISPLAY`? Remote rendering is not supported.

QLV5645-QLJ, 2020-02-06:
The simulator is running on the Server. And I set the DISPLAY value to be can local computer IP and Port. The log says this:
[image]

herb-kuta-lge, 2020-02-06:
>+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01 Driver Version: 440.33.01 CUDA Version: 10.2 |
+-----------------------------------------------------------------------------+

Are there really no GPUs listed by `nvidia-smi`? That would certainly be problem...

QLV5645-QLJ, 2020-02-06:
Actually. I have GPU on my remote container.
[image]

wangyixu14, 2020-02-06:
> > +-----------------------------------------------------------------------------+
> > | NVIDIA-SMI 440.33.01 Driver Version: 440.33.01 CUDA Version: 10.2 |
> > +-----------------------------------------------------------------------------+
> 
> Are there really no GPUs listed by `nvidia-smi`? That would certainly be a problem...

Oh, there are one gpu but I am using the remote desktop by google. 
So lgsvl doesn't support this?

Thanks

herb-kuta-lge, 2020-02-07:
>  using the remote desktop by google.

wangyixu14, 2020-02-08:
> > using the remote desktop by google.
> 
> Do you mean [Chrome Remote Desktop](https://chrome.google.com/webstore/detail/chrome-remote-desktop/gbchcmhmhahfdphkhkmpfmihenigjmpp)?

Yes, right.

herb-kuta-lge, 2020-02-08:
We have not tried to use Chrome Remote Desktop with the Simulator.

herb-kuta-lge, 2020-02-08:
There's this note at the top of https://cloud.google.com/solutions/chrome-desktop-remote-on-compute-engine:

> Note: This solution is not suitable for graphically intensive applications, including playing videos, because those typically require hardware graphics acceleration as well as a network that has high bandwidth and low latency. If you want to run graphically intense applications remotely, see the [Creating a virtual GPU-accelerated Linux workstation](https://cloud.google.com/solutions/creating-a-virtual-linux-workstation) tutorial for an alternative solution.

Simulator is certainly a graphically intensive application, so your best bet would be to try the alternative solution the note mentions.

wangyixu14, 2020-02-08:
Thanks for the reply.

closed 2020-02-08 by wangyixu14