lgsvl/simulator #623: can't add project in UnityHub when building the project

Allenhe123, 2020-02-05:
i just follow the Instructions to build standalone executable.
Step 7: In the Projects tab, click Add and select the folder that the Simulator was cloned to
There's no respond after i add the cloned simulator project.
I am using a Unity persional license.
Thanks.

Allenhe123, 2020-02-05:
I have found that someone had this issue before: #293.
I am using a UnityHub2.2.2 which is the latest version.
My platform is Ubuntu18.04LTS.

Allenhe123, 2020-02-05:
[info-log.txt](https://github.com/lgsvl/simulator/files/4160102/info-log.txt)

Attachment is the log file of UnityHub.

ntutangyun, 2020-02-05:
Here's my Setup: 
Unity Hub:  2.2.2
Unity: 2019.1.10f1
System: Ubuntu 18.04

Everything works fine so far, including build the simulator standlone executable. 
What's your Unity version @Allenhe123

Allenhe123, 2020-02-05:
My Unity version is: UnitySetup-2019.1.10f1 @ntutangyun

ntutangyun, 2020-02-06:
@Allenhe123 have you tried to check if you have following files as indicated by the `info-log.txt`, 
`/home/bstcd/.local/share/unity3d/prefs` ?

If cannot find, maybe you could try to reinstall the unity editor ? hopefully it works.

Allenhe123, 2020-02-06:
Thank you @ntutangyun 
Yes, i have double checked it and the unity3d/prefs directory was not exist.
Maybe the root cause is Unity3d was not installed successfully. I will try to reinstall it. :)

SeymurSm, 2020-02-06:
It happens on Ubuntu.  To overcome the problem create dummy project by using New button. Close newly created project and try to add simulator with Add button.
It should work

Allenhe123, 2020-02-07:
@SeymurSm Thank you for your advice.
I have resolved the issue: just create a new u3d project before add the simulator project.

closed 2020-02-07 by Allenhe123