lgsvl/simulator #624: How to edit (not build) the project in Unity Editor

ziranw, 2020-02-05:
Hi, it has been a long time since I used LGSVL from the last time, which I think was a version in 2018. I was able to load the whole project in Unity Editor and make customized changes.

However, after I downloaded the newest version (2020.01-rc1) source code, I wasn't sure how to operate it. I loaded the project in Unity editor but returned me a bunch of errors. I also didn't know where to find the map, or how could I load the map asset bundle.

For my research, I want to make some customized changes in the project (such as enabling vehicle-to-vehicle communication feature), so I have to make it through Unity Editor, not the executable file. I wasn't able to find the earlier version (source code) through LGSVL website as well.

mark-gerow-lge, 2020-02-05:
Hey @ziranw,

Our simulator has changed a lot since 2018, I recommend you follow our Quick Start guide here:

https://www.lgsvlsimulator.com/docs/#quick-start

If you find any errors working through that, please post them here and we can narrow down your problem. For getting maps/vehicles you should be able to find our sample content here:

https://content.lgsvlsimulator.com/

And use the links provided to download your maps of choice to your local simulator build through our Web UI as the documentation shows.

As for vehicle to vehicle communication you could probably implement this as a Sensor Plugin or Controllable without having to alter the source. You can read up on them here:

https://github.com/lgsvl/simulator/blob/master/Docs/docs/controllable-plugins.md

https://www.lgsvlsimulator.com/docs/sensor-plugins/

Feel free to follow up with any further questions.

-Mark Gerow

ziranw, 2020-02-05:
Hi Mark,

Thank you very much for the prompt reply!

Yes, I have gone through the quick start guide https://www.lgsvlsimulator.com/docs/build-instructions/ and was able to open the latest project (simulator-2020.01-rc1) in the Unity editor. However, after I opened that project, I wasn't able to find the scenes (such as BorregasAve) in the project path, where the only one under the "Scenes" folder is "LoaderScene". The console returned me 999+ errors.

I intent to play the project in the Unity Editor without using the WebUI. I think my major concern now is how to find/open the correct scene in the Unity Editor. Do I need to load anything into the "External" folder?

Thanks.

mark-gerow-lge, 2020-02-05:
Maps and Vehicles aren't included in the simulator project itself as we bundle them now. Some (like borregas) are available in built bundle form only, but others you can clone as a submodule into External/Maps or External/Vehicles then open them up in the Unity Editor and edit them as you need.

Example Vehicle:

https://github.com/lgsvl/Jaguar2015XE

Example Map:

https://github.com/lgsvl/SanFrancisco

ziranw, 2020-02-05:
Thanks Mark. Just out of curiousty, are any of the previous source-code versions (before introducing the WebUI or before 2019.5) available to download? The ones I can find online are all executable versions.

mark-gerow-lge, 2020-02-05:
Here is the last Pre-WebUI version we have IIRC, you will need Git LFS to clone properly.

https://github.com/lgsvl/simulator-2019.05-obsolete

ziranw, 2020-02-05:
> Here is the last Pre-WebUI version we have IIRC, you will need Git LFS to clone properly.
> 
> https://github.com/lgsvl/simulator-2019.05-obsolete

Thanks so much Mark!

closed 2020-04-08 by EricBoiseLGSVL