lgsvl/simulator #626: How to use the cyber bridge between Apollo 5.5 and LGSVL

mm-joseph, 2020-02-07:
Hi everyone! 
I am trying to connect Apollo 5.5 to LGSVL  simulator but I encountered some issues with the cyber_bridge module.
Do you have any idea on how to setup or adapt the bridge to Apollo 5.5?
Porting this module from Apollo 5.0 (simulation branch) to Apollo 5.5 seems to not work properly, the module establishes the connection with the simulator but it is not possible to run the simulation.

Setup info:
Apollo 5.5
OS: Ubuntu 18.04.3 LTS
x86

Thank you in advance.

EricBoiseLGSVL, 2020-02-08:
@mm-joseph We don't yet support Apollo 5.5 but we have it on our road map.  Hopefully soon.

mm-joseph, 2020-02-08:
Can I adapt the code by myself? Which are the basic files to change?

ntutangyun, 2020-02-09:
@mm-joseph you could check out the files changes for the Apollo 5.0 fork, from the latest commits back to the moment it's first forked.

lemketron, 2020-05-12:
> Can I adapt the code by myself? Which are the basic files to change?

[Several PRs](https://github.com/ApolloAuto/apollo/pulls?q=is%3Apr+author%3Arongguodong+%22LGSVL+Simulator%22) have been submitted to the ApolloAuto project to add support for LGSVL simulator, several of which have already been accepted. If anyone is eager to test this, you can check out Apollo master and and then manually add in the changes in the unmerged PRs.

We will provide updated documentation for using LGSVL Simulator with Apollo (master) soon.

rongguodong, 2020-05-18:
We have all five PRs merged into Apollo now. The latest Apollo master should works with our simulator. 
@mm-joseph @ntutangyun You can clone latest Apollo and test it. You can follow most of instructions to Apollo 5.0 with two changes:
- Call "bootstrap_lgsvl.sh" instead of "bootstrap.sh" to launch Dreamview; and
- Select  the `Mkz Lgsvl` mode, in addition to select `Lincoln2017MKZ LGSVL` vehicle and `BorregasAve` map.
Please let us know if you found any problems when using latest Apollo with LGSVL Simulator.

miluit01, 2020-05-20:
> We have all five PRs merged into Apollo now. The latest Apollo master should works with our simulator.
> @mm-joseph @ntutangyun You can clone latest Apollo and test it. You can follow most of [...]

@rongguodong: 

Hi Guodong, 

I checked LGSVL Simulator 2020.03 with Apollo 5.5. I observed the timestamps between Simulator and Apollo and found out they are not synchronized:

[image]

Left: apollo.drivers.gnss.Imu topic sent from Simulator
Right: apollo.common.monitor.MonitorMessage Apollo internal cyber time

The delta increases during whole time.

I investigate a little bit in source code:

Simulator takes a fix step time and increase the timestamp each loop with this delta, am I right?
Cyber takes time from system clock.

Issue:
This inconsistent timestampes causes a unexpected result of perception module of Apollo. The detected objects are not stable: position of object not updated, object cannot be detected, etc.

My quick and dirty workround:

To avoid this issue I adapted the source code on Apollo side:
in file modules/perception/onboard/transform_wrapper/transform_wrapper.cc:
line 219: cyber::Time query_time = cyber::Time(0);

in file /home/luming1/workspace/apollo/modules/perception/onboard/component/trafficlights_perception_component.cc
line 676: apollo::cyber::Time query_time(0);

After that I get "stable"  perception output.

Anyway from my point of view, the timestample needs to be synchronized between simulator and cyber. Otherwise simulation cannot provide any plausible result.

Thanks!

rongguodong, 2020-05-20:
Yes, we know the time difference between simulation time and real time. But I did not see any obvious issue caused by it. Can you describe more details of what are "unexpected result" of perception module? How can I reproduce this locally? And when you made those changes, how do you tell the results are more stable?

As for the timestamp, we are investigating several solutions, including using real time in simulator and using MOCK mode in Apollo.

rongguodong, 2020-05-24:
Per offline sync with @miluit01, the problem of "unstable perception" is because of wrong mode. You need to select "Mkz Lgsvl" mode (from the drop down menu left to the vehicle selection menu). With this mode in use, the perception module should work well.

miluit01, 2020-05-26:
@rongguodong Thanks for your classification! As we offline discussed, the core issue is the timestamp diff between simulator and cyber time. I hope that you will have a solution with Baidu developer soon. Many thanks!

zhanghming, 2020-05-26:
@rongguodong Hi, I'm using the LGSVL-linux64-2020.03, I can't find the "Mkz Lgsvl" mode menu in LGSVL interface,  is this menu in browser interface or the lgsvl app interface? 

What the default mode and if I didn't change this option, is it right or not?

lemketron, 2020-05-26:
> Hi, I'm using the LGSVL-linux64-2020.03, I can't find the "Mkz Lgsvl" mode menu in LGSVL interface, is this menu in browser interface or the lgsvl app interface?
> 
> What the default mode and if I didn't change this option, is it right or not?

@zhanghming The "setup mode" menu is found in Dreamview (in the browser), not in the LGSVL Simulator. It is to the left of the "vehicle" menu and should be set to "Mkz Lgsvl".
[image]

zhanghming, 2020-05-26:
@lemketron Thanks for your reply. 
What the Apollo repository are you use? I can not find the "Mkz Lgsvl" option in the Dreamview, there are only "Mkz Close Loop" and "Mkz Standard Debug". And I used the lgsvl/apollo-5.0 repository.

lemketron, 2020-05-26:
> What the Apollo repository are you use? I can not find the "Mkz Lgsvl" option in the Dreamview, there are only "Mkz Close Loop" and "Mkz Standard Debug". And I used the lgsvl/apollo-5.0 repository.

@zhanghming Sorry, it seems that @rongguodong and I were referring to the latest apollo (in the apollo master branch). 

For lgsvl/apollo-5.0 it should be "Mkz Standard Debug" as shown in the [LGSVL Simulator docs for Apollo 5.0](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/) and [screen shot](https://www.lgsvlsimulator.com/docs/images/full_size_images/apollo5-0.png).

zhanghming, 2020-05-28:
@lemketron  Thanks a lot, I ran all things successfully.

closed 2020-09-25 by lemketron