lgsvl/simulator #627: When the car stops at traffic light, it never start to drive

Martin20, 2020-02-08:
Hello,

I installed Apollo-5.0 on ubuntu 16.04 and I built it. Everything is fine except for two things:

1-The traffic light is turned off after a while in the LGSVL simulator and I think due to that, the car stops at intersections when the light is red and never starts to drive.

2- GPS also turns off itself after a short time. 

Any solutions?

EricBoiseLGSVL, 2020-02-12:
@Martin20 
1) Does the green light emission stop working or is it detection?  What map are you using?  We have an issue with apollo if you are not in a far right lane when you start, that it won't drive.  Place ego spawn in the far right lane and if it still won't drive press forward slightly to nudge it forward.  We are still trying to debug this.
2) @hadiTab do you know what would cause this?

ntutangyun, 2020-02-12:
@Martin20 when the ego car stops at junction under red light, when the light turns green, try manually driving it forward a bit. see if Apollo could manage to take over and start to drive again ?

if yes, then I remember someone has mentioned this issue and somebody has made some modifications on Apollo's end, to automaticlaly enable auto pilot when the light turns green.

Martin20, 2020-02-12:
Thank you for your reply.

The light is working but I think the detection is the problem. I used San Francisco, CubeTown, and BorregasAve. In all of them, after a while, the "traffic light" button in the simulator is turned off.  I am not sure it does not start to drive because of that but I think the car does not detect the traffic light. 

It might be because some parts of the apollo do not work fine. Why I am saying that? because when I run ./apollo.sh test, I receive the messages below. Do you know those messages are related to the traffic light?

By the way, can I fix the problems are showing the ./apollo.sh test?

Actually, I am working in school and I do not have the root password for the apollo machine. But, I have the sudo permission. By knowing that, is that possible I fix those problems showed in the messages below?

 WARNING: //modules/v2x/v2x_proxy/os_interface:os_interface_test: Test execution time (0.4s excluding execution overhead) outside of range for MODERATE tests. Consider setting timeout="short" or size="small".
//cyber/common:file_test                                                 FAILED in 0.4s
  /root/.cache/bazel/_bazel_root/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/cyber/common/file_test/test.log
//modules/dreamview/backend/simulation_world:simulation_world_service_test FAILED in 2.2s
  /root/.cache/bazel/_bazel_root/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/dreamview/backend/simulation_world/simulation_world_service_test/test.log
//modules/map/pnc_map:pnc_map_test                                       FAILED in 0.4s
  /root/.cache/bazel/_bazel_root/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/map/pnc_map/pnc_map_test/test.log
//modules/map/pnc_map:route_segments_test                                FAILED in 0.4s
  /root/.cache/bazel/_bazel_root/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/map/pnc_map/route_segments_test/test.log
//modules/perception/camera/test:camera_lib_lane_detector_darkscnn_lane_detector_test FAILED in 7.0s
  /root/.cache/bazel/_bazel_root/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/perception/camera/test/camera_lib_lane_detector_darkscnn_lane_detector_test/test.log
//modules/perception/camera/test:camera_lib_lane_postprocessor_darkscnn_lane_postprocessor_test FAILED in 5.6s
  /root/.cache/bazel/_bazel_root/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/perception/camera/test/camera_lib_lane_postprocessor_darkscnn_lane_postprocessor_test/test.log
//modules/planning/navi/decider:navi_decider_test                        FAILED in 2.0s
  /root/.cache/bazel/_bazel_root/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/planning/navi/decider/navi_decider_test/test.log
//modules/planning/tasks/deciders/open_space_decider:open_space_roi_decider_test FAILED in 1.1s
  /root/.cache/bazel/_bazel_root/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/planning/tasks/deciders/open_space_decider/open_space_roi_decider_test/test.log

Executed 522 out of 522 tests: 514 tests pass and 8 fail locally.
============================
[ERROR] Test failed!
[INFO] Took 1551 seconds
============================

Martin20, 2020-02-12:
I am able to do that. When I move forward with the keyboard, apollo takes
care of that. Do you know how and where would be modified in the backend?

best regards,

ntutangyun, 2020-02-14:
@Martin20 Back to [an old issue](https://github.com/lgsvl/simulator/issues/166). there is one guy mentioned that He was able to fix the problem.

All I know is that if Apollo stops due to traffic light before entering a junction, 
what you could do is check out this file  `modules/planning/traffic_rules/traffic_light.cc`. 

This file is the main code that tells Apollo to stop in front of red light.

Martin20, 2020-02-18:
It does not help me. Which part of the [old issue](https://github.com/lgsvl/simulator/issues/166) solves this problem?

ntutangyun, 2020-02-18:
on Jun 6, 2019,  yanbec replied:
> @ntutangyun I had this problem with 3.0 before, can't tell you how I fixed it, but it was a problem with traffic light recognition. On the right side in dreamview, does it show the recognized traffic light status or "unknown"? If unknown, it's supposed to stop.

do check out the file `modules/planning/traffic_rules/traffic_light.cc` if the stopping reason is due to traffic light. You could even comment out the following part to let the vehicle ignore the traffic light:

```c++
    // build stop decision
    ADEBUG << "BuildStopDecision: traffic_light["
           << traffic_light_overlap.object_id
           << "] start_s[" << traffic_light_overlap.start_s << "]";
    std::string virtual_obstacle_id =
        TRAFFIC_LIGHT_VO_ID_PREFIX + traffic_light_overlap.object_id;
    const std::vector<std::string> wait_for_obstacles;
    util::BuildStopDecision(virtual_obstacle_id,
                            traffic_light_overlap.start_s,
                            config_.traffic_light().stop_distance(),
                            StopReasonCode::STOP_REASON_SIGNAL,
                            wait_for_obstacles,
                            TrafficRuleConfig::RuleId_Name(config_.rule_id()),
                            frame, reference_line_info);
```

I don't have the solution either, besides commenting this part out...

hadiTab, 2020-02-18:
@Martin20, I've encountered the issue you are describing when the traffic light goes out of the view of the camera. That is, it initially detects the traffic light and begins to stop, but right at the stop location the traffic light is just a bit outside the view of the camera and it fails to detect it and just shows "unknown". Once you go forward a little bit, you will have entered the intersection at which point Apollo will drive regardless of the status of the traffic light.

Do you have the debug window open for the traffic light camera? If so check and see if you can view it. It's also possible that it only goes out of the view of the telephoto camera and needs to revert to using the 'normal' camera for which the traffic light detection model does not work as well.

Martin20, 2020-03-03:
Thank you @hadiTab. Actually, I turn the camera sensor inside the dreamview and see the image below:
[image]

Is that enough for the detection?

I run cyber_monitor and see the channel for the traffic_light and its module delay is less than 1 second in Module Delay panel. After a while, i.e. 2 minutes, the delay for traffic_light becomes increasing and also I cannot see the traffic_light channel updates in cyber_monitor. It seems something happens after some time. For example overflowing a buffer, a queue, etc.
Have you had something similar to this situation?

Two more questions, 

- how can I have the debug window for traffic_light camera? In dreamview as I said?
- How can I change the camera to normal?

Thanks.

closed 2020-10-06 by EricBoiseLGSVL

lemketron, 2020-10-06:
Our new [Modular Testing](https://www.lgsvlsimulator.com/docs/modular-testing/) feature (which uses ground truth from simulator instead of running Apollo perception and traffic light modules) is available in the 2020.06 release.

If you have not tried this, I would encourage you to take a look since it makes it much easier to run Apollo and LGSVL Simulator on a single machine by publishing ground truth information for perception obstacles as well as traffic light information, and means you don't have to worry about camera issues for traffic light detection (unless that's something that you're specifically wanting to use or test).

This works with Apollo 5.0 and Apollo master.

Martin20, 2020-10-21:
I need to have several vehicles. Hence, I need to run it on several machines. If you have a solution to run several vehicles on the same machine, I can do your suggestion.

Do you have?

EricBoiseLGSVL, 2020-10-21:
If you have minimal sensors with what @lemketron suggested you can run multiple vehicles on one machine.

Martin20, 2020-10-21:
How is it possible when for adding a new vehicle we need a distinct IP address?

EricBoiseLGSVL, 2020-10-21:
You can set different connection addresses in the PythonAPI and in WebUI

Martin20, 2020-11-06:
@EricBoiseLGSVL Do you have any tutorial for having several vehicles through PythonAPI?

EricBoiseLGSVL, 2020-11-07:
@Martin20 I'll see if I can find one we use for testing and post here.  I'll also add it to the next release for anyone else that might want it.  Thanks for the good request

Martin20, 2020-11-07:
@EricBoiseLGSVL Thank you very much for your help.

EricBoiseLGSVL, 2020-11-09:
@Martin20 I spoke with another engineer on adding multiple ego vehicles, it is the same as adding one. Just declare the other vehicle like the first one and make sure the spawn position is not the same.  I will try to create an example later this week

Martin20, 2020-11-10:
> @Martin20 I spoke with another engineer on adding multiple ego vehicles, it is the same as adding one. Just declare the other vehicle like the first one and make sure the spawn position is not the same. I will try to create an example later this week

Thank you. I will wait for that.