lgsvl/simulator #629: How to display point cloud of lidar in unity editor?

ghoulich, 2020-02-12:
Simulator version: 2020.01
Unity version: 2019.1.10f1

I created a lidar sensor and two walls in the SingleLaneRoad evironment, cloned from:
https://github.com/lgsvl/SingleLaneRoad.git

When I pressed Play button in the Unity Editor, although the lidar was rotating normally and points data could be obtained via debugging and no errors happened, point cloud still couldn't be displayed in neither Scene nor Game tab!

How can I build such scenario with one lidar and two walls, then display point cloud correctly?

Here is my configuration:
[image]

PS: above scenario can work well in version 2019.05

EricBoiseLGSVL, 2020-02-12:
@ghoulich do you have any errors?  Also, the walls need HDRP/Lit shader/mat applied.  You also need to enable the sensor visualization in the UI.

ghoulich, 2020-02-13:
> @ghoulich do you have any errors? The walls need HDRP/Lit shader/mat applied and need to enable the sensor visualization in the UI.

No errors happened during Playing, I will try your solution later, thanks a lot!

ghoulich, 2020-02-13:
> @ghoulich do you have any errors? The walls need HDRP/Lit shader/mat applied and need to enable the sensor visualization in the UI.

I have checked my scenario configuration, walls already have HDRP/Lit set in DefaultHDMaterial:
[image]

And when the scenario entered Play mode, I cannot find any sensor visualizer toggle in Simulator UI, toggle list is empty:
[image]

How can I do next? Shall I make a tarball of this scenario and send it to you?

EricBoiseLGSVL, 2020-02-13:
@ghoulich The issue you have is that you are not spawning your new lidar sensor (actually no sensors and no vehicle)

You need a vehicle to run the simulator.  It does not support running just a sensor at this time.

You need to [configure](https://www.lgsvlsimulator.com/docs/sensor-json-options/) the sensors in the WebUI so the Simulation knows to spawn it and the parameters that it needs.  Run Loader scene, open webui, add to vehicle config.  It will save the config to the vehicle and when you select it in Editor, Simulator->Development Settings... you select that vehicle and it loads the sensors you list in json

You need to also add the prefab to the sensors list in Assets/Resources/RuntimeSettings SensorPrefabs list.  This will conflict with new sensors we create and push so you should build this as a sensor plugin so the simulation.  See [here](https://www.lgsvlsimulator.com/docs/sensor-plugins/)

Hope this helps and let us know if you have issues.

ghoulich, 2020-02-13:
> @ghoulich The issue you have is that you are not spawning your new lidar sensor (actually no sensors and no vehicle)
> 
> You need a vehicle to run the simulator. It does not support running just a sensor at this time.
> 
> You need to [configure](https://www.lgsvlsimulator.com/docs/sensor-json-options/) the sensors in the WebUI so the Simulation knows to spawn it and the parameters that it needs. Run Loader scene, open webui, add to vehicle config. It will save the config to the vehicle and when you select it in Editor, Simulator->Development Settings... you select that vehicle and it loads the sensors you list in json
> 
> You need to also add the prefab to the sensors list in Assets/Resources/RuntimeSettings SensorPrefabs list. This will conflict with new sensors we create and push so you should build this as a sensor plugin. See [here](https://www.lgsvlsimulator.com/docs/sensor-plugins/)
> 
> Hope this helps and let us know if you have issues.

Do you mean that old version (2019.05) can support this scenario, but current version (2020.01) cannot support? Is this the reason why I can get PointCloud data in debugging, but nothing displayed?

EricBoiseLGSVL, 2020-02-13:
@ghoulich Excellent, looks great.  Yes, at this time sensors are bound to vehicles.  We are looking into a way to support sensors sans a vehicle  but not supported at this time.  You could try creating a [controllable](https://www.lgsvlsimulator.com/docs/controllable-plugins/) that communicates with a custom sensor that would then send data through the bridge.

ghoulich, 2020-02-14:
> @ghoulich Excellent, looks great. Yes, at this time sensors are bound to vehicles. We are looking into a way to support sensors sans a vehicle but not supported at this time. You could try creating [...]

Thanks for your help!
I will keep track of the version update and look forward to supporting the independent sensor as soon as possible!

closed 2020-02-14 by ghoulich