lgsvl/simulator #630: how can I get pointcloud ring

neophack, 2020-02-12:
[image]
[image]
[image]

lgsvl have no ring in data

EricBoiseLGSVL, 2020-02-12:
@hadiTab is ring supported in our msg to rviz?

closed 2020-10-20 by EricBoiseLGSVL

soldierofhell, 2021-01-22:
@EricBoiseLGSVL, @hadiTab, I've inspected the ROS2 bridge code and indeed, there's no ring field in PointCloud2 message. Are there any plans to extend it? For Velodyne-type rotating lidars this is very important, e.g. some SLAM algorithms use it directly

bahattinkoc, 2021-02-27:
Any progress? i still can't get the ring data?