lgsvl/simulator #631: Pytohn API run() doesn't start the simulation

joshwapohlmann, 2020-02-13:
Using 2020.01-rc1 the Python API function `run()` doesn't actually start the simulation. manual press on the "play" button is still required.

## Steps to reproduce
1. Start simulator
2. Run simulation in "API only" mode
3. Connect via python API
4. load a scene and issue `run()` command.

## Expected behavior
The simulator should start on it's own.

## Observed behavior
It doesn't

## Additional Notes
When `reset()`ing the scene, it will not pause the simulation and it works.

EricBoiseLGSVL, 2020-02-13:
@shalinmehtalgsvl were you having this issue?

EricBoiseLGSVL, 2020-02-13:
@joshwapohlmann this is a bug that we have on the list to fix.  Thanks for letting us know

joshwapohlmann, 2020-02-13:
@EricBoiseLGSVL I just did some quick digging and checked the implementation of the run command against the actual button press. 
The difference I saw was the usage of some kind of `timeScaleLock`, which is not used in any way by the run() implementation. Just a quick guess/hint, maybe of help.

EricBoiseLGSVL, 2020-02-13:
@joshwapohlmann excellent, that helps.

closed 2020-04-08 by EricBoiseLGSVL