lgsvl/simulator #633: Simulator crashed from "LGSVL Simulator Configuration"

kanbei, 2020-02-13:
Hello!
I press the OK button in "LGSVL Simulator Configuration", it suddenly crashed.
My graphics card is GeForce RTX 2070.
According to the Player.log, it seems to be the drive of graphics card is not be supported.
My driver version is  430.64, cuda version is 10.0, and I have tried different version of the drive and cuda, the simulator still crashed.
What's the reason of it?

--------------------------------------------------------------------------------
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=2
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan error VK_ERROR_INCOMPATIBLE_DRIVER (-9) file: ./Runtime/GfxDevice/vulkan/VKContext.cpp, line: 333
Vulkan error./Runtime/GfxDevice/vulkan/VKContext.cpp:333
Vulkan detection: 0
No supported renderers found, exiting
 
(Filename: ./PlatformDependent/LinuxStandalone/main.cpp Line: 639)

Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox

[image]

EricBoiseLGSVL, 2020-02-13:
@kanbei Have you updated your driver?  Any other error logs you can post?

EricBoiseLGSVL, 2020-02-13:
@kanbei Also, if you are running Apollo 5.0, it does not support Volta and Turing architecture GPUs (Titan V and RTX GPUs)  To run simulator our engineer that has a RTX 2080Ti and his driver is 440.59 and Cuda 10.2

ck-leo, 2020-02-26:
@EricBoiseLGSVL I have the same issue, My graphics card is GeForce RTX 2060 6G

the error file is :
Desktop is 1920 x 1080 @ 144 Hz
[Vulkan init] extensions: count=1
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
Vulkan error VK_ERROR_INCOMPATIBLE_DRIVER (-9) file: ./Runtime/GfxDevice/vulkan/VKContext.cpp, line: 333
Vulkan error./Runtime/GfxDevice/vulkan/VKContext.cpp:333
Vulkan detection: 0
No supported renderers found, exiting
 
(Filename: ./PlatformDependent/LinuxStandalone/main.cpp Line: 639)

Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox

And my nvidia-smi result is :

[image]

Is there some other suggestions?Thanks.

EricBoiseLGSVL, 2020-02-26:
@ck-leo Please see if this [issue 287](https://github.com/lgsvl/simulator/issues/287) helps.

ck-leo, 2020-02-27:
> 抄　送：ck-leo <ck_leo@aliyun.com>; Mention <mention@noreply.github.com>
> 主　题：Re: [lgsvl/simulator] Simulator crashed from "LGSVL Simulator Configuration" (#633)

OK, Thanks.

——
------------------------------------------------------------------

EricBoiseLGSVL, 2020-04-08:
@kanbei @ck-leo We have a new release using a newer version of Unity and should be out in a few days.  Unity update fixes many Vulkan issues but if not please re-open this issue

closed 2020-04-08 by EricBoiseLGSVL