lgsvl/simulator #635: Map Signal IDs doesn't match the Apollo 5.0 base_map

ntutangyun, 2020-02-14:
Hi team,

I found that the signals in lgsvl simulator do not match those in Apollo 5.0 base_map in terms of IDs.

Take `borregas_ave` map for example:
below is the picture generated from Apollo 5.0 base_map file, where the traffic signals are drawn for the first junction:

[image]

Note that the traffic signal IDs are simplified, i.e. from `signal_2` to `s_2` etc. 

[image]

When the ego vehicle approaches as shown above, there are 4 signal associated.
According to Apollo 5.0 base map, they are `signal_2`, `signal_5`, `signal_6` and `signal_12`. This also matches the output taken from `/apollo/perception/traffic_light` default camera data are fed to Apollo 5.0.

However, when I was trying to get the signals from lgsvl scripts, with following code:
```c#
closestLane = SimulatorManager.Instance.MapManager.GetEgoCurrentLane(transform.parent.transform.position);
Debug.Log("====================================================");
Debug.Log("there are " + closestLane.stopLine.signals.Count + " signals");
foreach (MapSignal signal in closestLane.stopLine.signals)
{
       Debug.Log(signal.ID);
       Debug.Log("-----");
}
```
this is what I got:

[image]

[image]

I'm not sure why the IDs are set this way. But they don't match with Apollo 5.0 base_map in the latest release. 

Or is it matching those in Autoware maps ?

Thanks.

EricBoiseLGSVL, 2020-02-14:
@luqiang21 or Tae know what is happening here?

cyberphysicalrobotics, 2020-02-14:
@ntutangyun Which Simulator version are you using?

cyberphysicalrobotics, 2020-02-14:
@ntutangyun There is id property, not ID. I think id property will have signal id. So please add checking id in your debug routine. Let me know your result.

ntutangyun, 2020-02-15:
@cyberphysicalrobotics Thank you for reply. 
I'm using the latest version. pulled from master branch, quite recently (1 Feb).

```c#
closestLane = SimulatorManager.Instance.MapManager.GetEgoCurrentLane(transform.parent.transform.position);
            Debug.Log("====================================================");
Debug.Log("there are " + closestLane.stopLine.signals.Count + " signals");
foreach (MapSignal signal in closestLane.stopLine.signals)
{
      Debug.Log("ID: " + signal.ID + " id: " + signal.id + " UID: " + signal.UID);
      Debug.Log("-----");
}
```

Below is what I got:
[image]

There are four related fields for `MapSignal` class, `string UID`, `uint ID`, `string id` and `string Key => UID`.  since Key is same as UID, so I just logged the other three. 

and Only `ID` is set, to a seemingly random number, the small `id` and `UID` are NULL.

closed 2020-04-25 by ntutangyun