lgsvl/simulator #636: Vehicle don't move in Apollo 3.5

simon13579, 2020-02-16:
ello,

I follow the instructions to build LGSVL Apollo 3.5 code and use the released simulator 2019.12, the route is generated but the vehicle don't move.
[image]

The error "RAW IMU message delay" and "RAW GPS message delay" displays in the dreamviewer, is it the root cause and how can I fix it? And sometimes the Perception in the module control tab is deselected but that doesn't happen every time . 
By the way, in the Routing Editing tab of the dreamviewer, I can't find the command "submit the routine", but only "send the routine request" can be found.

Thanks,
Simon

ntutangyun, 2020-02-16:
@simon13579 try use keyboard to manually drive the vehicle a little bit forward and see if the vehicle starts moving ?

simon13579, 2020-02-16:
By using keyboard, the vehicle starts moving, but it still can't follow the routine. It seems that the apollo software doesn't send control control command to simulator.

ntutangyun, 2020-02-16:
@simon13579 the default mode should be correct.

By right after you drove the vehicle a little bit forward manually, Apollo should be able to take over and continue to drive it following the route. 

by your screenshot, there is a bus at the intersection, but in dreamview, Apollo's perception module is not capturing any obstacles. This is not right. 

1. Maybe you could give the system a little bit time to start up. And see if Apollo is able to start drawing those obstacles in the Dreamview page.

2. In Dreamview -> tasks page, there should be indicators of message delays of various topics. You could check if any module is not sending messages. 

3. you could get into the docker container, and use `cyber_monitor` to check if all the modules are working correctly. 

4. if some module is not working, you could get into the docker container, and try start the module manually. (note turn off the same module in Dreamview first). e.g. `cyber_launch start modules/perception/launch/percetion.launch` etc.   

Above is based on my understanding of Apollo 5.0. for Apollo 3.5 please adjust accordingly. 

Hope it helps.

simon13579, 2020-02-17:
After I use cyber_monitor, it shows that the perception module is not working. Then I use the command cyber_launch start modules/perception/production/launch/perception.launch
[image]
[image]

Only perception/motion_service starts working. And I check the perception log file in the data/log folder, no error information is shown.
[image]

Where can I find the error information of perception module?
And I don't install opencv, caffe2 and CUDA on my host computer, is that OK?

Best Regards.

ntutangyun, 2020-02-17:
@simon13579 could you check if the simulator is publishing any lidar sensor output. i.e. `point_cloud` related topics ?

Not sure about Apollo 3.5, but for Apollo 5.0, the perception module requires `point_cloud` data to output its own `/apollo/perception/obstacles` message.

simon13579, 2020-02-17:
@ntutangyun could you please tell me how to check? Sorry, I just start to use LGSVL and Apollo. Thanks!

simon13579, 2020-02-22:
There is no point_cloud topics inside cyber_monitor, only camera related topic. Is it correct that simulator only output the camera message?
[image]

And I check some document of Apollo 3.5, it seems that its perception don't requires lidar data.
In dreamviewer the warning message "Raw IMU message delay" and "Raw GPS message delay“ are shown, could it be the root cause? How to fix this issue? My graphic card is GTX 1660ti that has 6G memory, could it be a problem?
[image]

Finally how can I find the root cause why perception module is not working? Where to find the log message? I check the perception log file in the data/log folder, no error information is shown.

Best Regards

ajay1606, 2020-04-06:
@simon13579 Did you fixed this issue? Am also getting the same warning message "Raw IMU message delay" and "Raw GPS message delay“ !

Could you please help me if you have already got fixed this issue?

Best Regards.

simon13579, 2020-04-25:
@ajay1606 No, this issue is still here, please let me know if you have any solution, thanks!

Best Regards.

EricBoiseLGSVL, 2020-10-20:
I'm guessing this issue is a 6gb GPU running simulator and Apollo.  This is not ideal.  I also recommend using Apollo 5.0 or current Apollo.  We have been working with the team to fix many issues with Apollo and LGSVL Simulator and we recommend using the latest version.

closed 2020-10-20 by EricBoiseLGSVL