lgsvl/simulator #637: Added trees are missing in LIDAR detection

riveranb, 2020-02-16:
I tried to add some tree (made from MTree, https://assetstore.unity.com/packages/tools/modeling/mtree-tree-creation-132433)

Everything looked fine but When I check LIDAR detection, I found out that trees are not scanned correctly.

Trees are using shaders written by MTree package, if I change shader to "HDRP/Lit" this issue solved.
So I want to ask about what if I writes my own shader or using 3rd party customized shaders, how to prepare for LIDAR detection?

EricBoiseLGSVL, 2020-02-18:
@riveranb Yes, you need to look at our changes to HDRP that we include in the source code to see how we enable LiDAR detection.  It is in the Packages folder on the root.  Normally this is just a JSON file of packages and versions but we move it to apply shader changes we need.  You can create your own but it still may need modification in HDRP.  Be aware that Unity makes changes to HDRP and can break when updating versions.

riveranb, 2020-02-23:
> @riveranb Yes, you need to look at our changes to HDRP that we include in the source code to see how we enable LiDAR detection. It is in the Packages folder on the root. Normally this is just a JSON [...]

Though you are right, you had opened source codes there. But there are just too many sources to read/research (HDRP + LGSVL parts).
However, after several days trying, finally I found clues in generated HDRP/Lit.shader codes.
After I add shader passes references from generated HDRP/Lit.shader, passes about Lidar-pass, sematic-pass, depth-pass.
Then my shader works in LGSVL simulator runtime! That's really good news to me.

Here is my shader modification for LGSVL simulator passes
`
		Pass
		{
			Name "SimulatorLidarPass"
			Tags { "LightMode" = "SimulatorLidarPass" }

			HLSLPROGRAM

			#define ATTRIBUTES_NEED_TEXCOORD0
			#define VARYINGS_NEED_POSITION_WS
			#define VARYINGS_NEED_TEXCOORD0

			#include "Packages/com.unity.render-pipelines.core/ShaderLibrary/Common.hlsl"
			#include "Packages/com.unity.render-pipelines.core/ShaderLibrary/Wind.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/ShaderLibrary/ShaderVariables.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/ShaderPass/FragInputs.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/ShaderPass/ShaderPass.cs.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Lit/LitProperties.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Material.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Lit/Lit.cs.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/ShaderPass/VaryingMesh.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Lit/LitData.hlsl"

			#pragma vertex Vert
			#pragma fragment Frag

			#include "Assets/Shaders/LidarPass.hlsl"

			ENDHLSL
		}

		Pass
		{
			Name "SimulatorSemanticPass"
			Tags { "LightMode" = "SimulatorSemanticPass" }

			HLSLPROGRAM

			#define ATTRIBUTES_NEED_TEXCOORD0
			#define VARYINGS_NEED_TEXCOORD0

			#include "Packages/com.unity.render-pipelines.core/ShaderLibrary/Common.hlsl"
			#include "Packages/com.unity.render-pipelines.core/ShaderLibrary/Wind.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/ShaderLibrary/ShaderVariables.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/ShaderPass/FragInputs.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/ShaderPass/ShaderPass.cs.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Lit/LitProperties.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Material.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Lit/Lit.cs.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/ShaderPass/VaryingMesh.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Lit/LitData.hlsl"

			#pragma vertex Vert
			#pragma fragment Frag

			#include "Assets/Shaders/SemanticPass.hlsl"

			ENDHLSL
		}

		Pass
		{
			Name "SimulatorDepthPass"
			Tags { "LightMode" = "SimulatorDepthPass" }

			HLSLPROGRAM

			#define ATTRIBUTES_NEED_TEXCOORD0
			#define VARYINGS_NEED_POSITION_WS
			#define VARYINGS_NEED_TEXCOORD0

			#include "Packages/com.unity.render-pipelines.core/ShaderLibrary/Common.hlsl"
			#include "Packages/com.unity.render-pipelines.core/ShaderLibrary/Wind.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/ShaderLibrary/ShaderVariables.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/ShaderPass/FragInputs.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/ShaderPass/ShaderPass.cs.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Lit/LitProperties.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Material.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Lit/Lit.cs.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/ShaderPass/VaryingMesh.hlsl"
			#include "Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Lit/LitData.hlsl"

			#pragma vertex Vert
			#pragma fragment Frag

			#include "Assets/Shaders/DepthPass.hlsl"

			ENDHLSL
		}
`

EricBoiseLGSVL, 2020-04-08:
@riveranb glad you were able to find a solution.  We hope to make this easier to use in the future

closed 2020-04-08 by EricBoiseLGSVL