lgsvl/simulator #638: Traffic Light Recognition with Autoware and LGSVL

pedroexenberger, 2020-02-17:
Hi everyone,

I've been setting up LGSVL simulator (2020.01) alongside autoware (1.12.0) and it has been working great. I'm incrementally making autoware modules work (and vizualizing them) using LGSVL simulated data as input. So far I can detect vehicles (with both ssd detection and lidar), and proceed with the navigation with the planner and motion modules (altough the ego-car hits other cars sometimes,but that is a future's problem).

However, I could not make the traffic light detection work when using autoware and LGSVL.
I've created a [question](https://answers.ros.org/question/344145/instructions-to-get-traffic-light-detection-working/) on ROS Answer with more details of the problem. But I'm not sure if the problem relies on autoware or LGSVL data, so I'm questioning here, from the LGSVL point of view. 

I'm using latest map data (testing with BorregasAve) from[Autoware-data repo](https://github.com/lgsvl/autoware-data), commit 05658fee453454af1f9290e0a0f76d4f19d1bb89. Does the vector map ont that repo contains information regarding traffic light position? I know that it is important for autoware to detect traffic lights. Also, autoware-data launch files contains calibration, which I'm running with the .launch files.

To start the traffic light recognition, I start both 

```
  <!-- traffic light recognition -->
  <!-- feat_proj -->
  <include file="$(find trafficlight_recognizer)/launch/feat_proj.launch" />

  <!-- region_tlr -->
  <include file="$(find trafficlight_recognizer)/launch/traffic_light_recognition.launch">
    <arg name="light_src" value="/traffic_signal_info" />
  </include>
```

Also, when I start the my_map.launch, I get

```
[ERROR] [1581679399.246782702]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246826716]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246842120]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246857334]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246870597]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246884724]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246899355]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246913409]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246926483]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246938762]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246950858]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246963482]: [createSignalMarkerArray] failed createPoleMarker: 1,1,6.75,0.4
[ERROR] [1581679399.246974994]: [createSignalMarkerArray] failed createPoleMarker: 2,2,6.75,0.4
[ERROR] [1581679399.246985849]: [createSignalMarkerArray] failed createPoleMarker: 2,2,6.75,0.4
[ERROR] [1581679399.247000304]: [createSignalMarkerArray] failed createPoleMarker: 2,2,6.75,0.4
[ERROR] [1581679399.247013807]: [createSignalMarkerArray] failed createPoleMarker: 2,2,6.75,0.4
[ERROR] [1581679399.247024335]: [createSignalMarkerArray] failed createPoleMarker: 2,2,6.75,0.4
[ERROR] [1581679399.247031686]: [createSignalMarkerArray] failed createPoleMarker: 2,2,6.75,0.4
[ERROR] [1581679399.247038977]: [createSignalMarkerArray] failed createPoleMarker: 2,2,6.75,0.4
[ERROR] [1581679399.247046151]: [createSignalMarkerArray] failed createPoleMarker: 2,2,6.75,0.4
[ERROR] [1581679399.247053824]: [createSignalMarkerArray] failed createPoleMarker: 2,2,6.75,0.4
[ERROR] [1581679399.247060915]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247068074]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247075010]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247081801]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247088545]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247095351]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247102208]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247108825]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247115573]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247122578]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247129414]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247137791]: [createSignalMarkerArray] failed createPoleMarker: 3,3,6.75,0.4
[ERROR] [1581679399.247144774]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247151518]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247158240]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247165703]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247172804]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247179539]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247186547]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247193345]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247200500]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247207424]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247214271]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
[ERROR] [1581679399.247220945]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
```

Can those errors be affecting the traffic light recognition? I saw in another issue https://github.com/lgsvl/simulator/issues/411#issuecomment-538521227 that it should not impact the overall functioning of autoware. So.. should the traffic light recognition be simple to perform? Am I missing some LGSVL configuration for enable that?

Thank you very much for your help.

EricBoiseLGSVL, 2020-02-18:
@luqiang21 does autoware support this?

cyberphysicalrobotics, 2020-02-19:
@pedroexenberger Which map were you using?

pedroexenberger, 2020-02-19:
@cyberphysicalrobotics I'm testing with BorregasAve map. I also tried setting up the San Fran. map (to check if it was a map issue), but it was too slow because the map is too large and I'm running both LGSVL and autoware in the same host. The single lane map does not have traffic lights. Haven't tried with other maps.
Thank you for the help.
Cheers.

EricBoiseLGSVL, 2020-02-20:
@pedroexenberger cubetown has lights to check.

pedroexenberger, 2020-02-20:
@EricBoiseLGSVL Thank you for the input! I'll give it a try and report the outcome later.

pedroexenberger, 2020-02-20:
@EricBoiseLGSVL  and @cyberphysicalrobotics, thank you for the help so far. I tried proceeding with the traffic light detection in the cubetown map, as suggested. Unfortunately it also didn't work. 

I also noticed that the /roi_signal topic is not even there. 

[image]

Also, CubeTown/my_launch/**my_detection.launch** starts
```
  <!-- region_tlr -->
  <include file="$(find trafficlight_recognizer)/launch/traffic_light_recognition.launch">
    <arg name="light_src" value="/traffic_signal_info" />
  </include>
```

In turn, autoware.ai/install/trafficlight_recognizer/share/trafficlight_recognizer/launch/**traffic_light_recognition.launch** starts

```
<node pkg="trafficlight_recognizer" type="region_tlr" name="traffic_light_recognition" output="log">
    <param name="image_raw_topic" type="str" value="$(arg camera_id)$(arg image_src)"/>
    <param name="camera_light_color_topic" type="str" value="$(arg camera_id)$(arg camera_light_src)"/>
</node>
```
autoware.ai/src/autoware/core_perception/trafficlight_recognizer/nodes/region_tlr/**region_tlr.cpp** subscribes to 4 topics, but only 2 are available:

```
	ros::Subscriber image_sub = n.subscribe(image_topic_name, 1, image_raw_cb);
	ros::Subscriber position_sub = n.subscribe("/roi_signal", 1, extractedPos_cb);
	ros::Subscriber tunedResult_sub = n.subscribe("/tuned_result", 1, tunedResult_cb);
	ros::Subscriber superimpose_sub = n.subscribe("/config/superimpose", 1, superimpose_cb);
```

[image]

So I proceed investigating, looking on feat_proj log files, to check the output during it's execution and I discover that the feat_proj never finish loading the vector map and keep stuck in this loop 

```
/* wait until loading all vector map is completed */
	ros::Rate wait_rate(100);
	std::cout << "Loading Vector Map. Please wait";
	while (vmap.points.empty() || vmap.lines.empty() || vmap.whitelines.empty() ||
	       vmap.lanes.empty() || vmap.dtlanes.empty() || vmap.vectors.empty() || vmap.signals.empty())
	{
		ros::spinOnce();
		std::cout << ".";
		wait_rate.sleep();
	}
```
https://gitlab.com/autowarefoundation/autoware.ai/core_perception/-/blob/1.12.0/trafficlight_recognizer/nodes/feat_proj/feat_proj.cpp#L422

And hence never proceeding to create the publisher for the /roi_signal. 

Are those vmap data collected from the autoware-data maps? I'm just start using autoware and LGSVL for a couple weeks, so I'm not very familiar with where does this data come from, but I think it's important that this data exists for traffic light to work, right?

I also noticed that the master branch in autoware (which is >= version 1.13.0, these .cpp files are modifyied). Do you know if is it a problem to be using version 1.12.0?

Again, thank you.

EricBoiseLGSVL, 2020-04-08:
@pedroexenberger We have it on our roadmap to support latest versions of Autoware soon.  We will look into this further

EricBoiseLGSVL, 2020-05-08:
@cyberphysicalrobotics Did you have any idea why traffic detection is not working?

pedroexenberger, 2020-05-08:
Hi @EricBoiseLGSVL, I just saw you mentioned this issue. When I was trying to fix https://github.com/lgsvl/simulator/issues/653 I tried a lot of different configurations to make that issue work and when I solved the traffic light was working as well. Unfortunately, I don't have a track of what was the exact cause or solution (because it was solved 'unawarely').

Autoware's traffic light detection is not very good (sometimes it gives me the wrong output and not always find the light pole), but LGSVL is not the problem. I can close the issue if that is sufficient for you.

EricBoiseLGSVL, 2020-05-08:
@pedroexenberger I'll close and we can reopen if we find out the root cause.  Thanks for the help, it might help another user

closed 2020-05-08 by EricBoiseLGSVL