lgsvl/simulator #639: Apollo 5.0 Gomentum Map lane missing

ntutangyun, 2020-02-18:
Hi team,

Just found that for [Apollo 5.0 Gomentum Map](https://github.com/lgsvl/apollo-5.0/tree/simulator/modules/map/data/gomentum),

there is `lane_238` in base_map.bin,  but there is no `lane_238` in the routing_map.bin.

I've used javascript to parse the protobuf binaries. Kindly check and let me know if I'm wrong. Thanks.

EricBoiseLGSVL, 2020-02-18:
@ntutangyun our importer might have simplified the lane data to work better in Simulator.  We can check, thanks

EricBoiseLGSVL, 2020-02-18:
@luqiang21 did we update Gomentum after our importer fixes?

ntutangyun, 2020-02-23:
I have found another lane data mismatch:
(Latest map files are pulled just before this comment)

in borregas_ave's the traffic light junction: 

[image]

It's clear that `lane_41` should be successor of `lane_17`. 
However this relation is missing from both `base_map.bin` and `routing_map.bin`.

below are screenshots from `base_map.bin` parsed by javascript:

[image]

[image]

> @ntutangyun our importer might have simplified the lane data to work better in Simulator. We can check, thanks
@EricBoiseLGSVL May I know if above issue is related to the simplification you mentioned earlier ? 
Thank You.

ntutangyun, 2020-02-26:
I've searched all the junction lanes (those connecting lanes that belong to junctions), 
* junction one, `lane_41` has no predecessor lane, 
* junction two, `lane_48` has no predecessor lane. 

and they are quite similar cases, both are filter lanes of right turning. 
Kindly help check it out.  thank you.

luqiang21, 2020-03-23:
@EricBoiseLGSVL I think we updated the map.
@ntutangyun Sorry for late reply. I am not sure why `lane_238` is missing since our simulator is providing only the file `base_map.bin`, `routing_map.bin` is generated inside apollo docker. You may want to check more in apollo side.

For junction lanes missing predecessor lane problem, I am guessing it is because some lanes are having wrong `laneTurnType` and are skipped when computing the relations. Can you try to comment out lines `684-688` https://github.com/lgsvl/simulator/blob/cb937deb8e633573f6c0cc76c9f451398b8b9eff/Assets/Scripts/Editor/ApolloMapTool.cs#L684, then generate the `base_map.bin` file? These lines are added to avoid wrong relations computed for `self-reverse lane`.

ntutangyun, 2020-03-24:
@luqiang21 Thanks for your kind pointer. Will check it out and update later. Closing for now.

closed 2020-03-24 by ntutangyun

ntutangyun, 2020-04-25:
Hi This is to update that after commenting out the lanes `684-688` in `simulator/Assets/Scripts/Editor/ApolloMapTool.cs`, the predecessor issue is indeed solved. 

Thanks for the help.

ntutangyun, 2020-04-30:
I think I've found out why the lane_238 is missing in routing_map. 

Below is the code snippet used to add lanes from base_map to routing_map, taken from 
`modules/routing/topo_creator/graph_creator.cc`

```c++
    if (forbidden_lane_id_set_.find(lane_id) != forbidden_lane_id_set_.end()) {
      ADEBUG << "Ignored lane id: " << lane_id
             << " because its type is NOT CITY_DRIVING.";
      continue;
    }
    if (lane.turn() == hdmap::Lane::U_TURN &&
        !IsValidUTurn(lane, min_turn_radius)) {
      ADEBUG << "The u-turn lane radius is too small for the vehicle to turn";
      continue;
    }
```

it has two criteria, 
* lane type needs to be CITY DRIVING 
* if lane turn is U-TURN, the turn radius (estimated by the first point, middle point, and the last point) should be larger than the vehicle's `min_turn_radius` configuration. 

`lane_238` is CITY_DRIVING lane and it happens to be a U-turn. Thus I think it's the turning radius.