lgsvl/simulator #640: Autoware Vector Map doesn't work in op_planner

mehmetdogru, 2020-02-18:
Hi,

The vector maps created for autoware doesn't work with "Autoware 1.13 -> op_planner"
Have you tried using these vector maps in op_planner?

Op_planner checks version (version 1 and version 2) for vector maps. When the vector maps data provided by lgsvl (autoware-data) is loaded, op_planner says it is version 2 vector map however when it reads the data it gives a segmentation error at the end of the "lane.csv" file. Because the last value of "FNID" in "lane.csv" doesn't exist in "node.csv" as a value of "NID". It is same for every vector map data provided by LGSVL in "lgsvl/autoware-data" repo.

OS : Ubuntu 18.04.4
Autoware Version: 1.13

Do you have any idea what the problem can be?
Thanks a lot in advance.

_Mehmet Doğru_

EricBoiseLGSVL, 2020-02-27:
@luqiang21 
Do you know what could be the issue here?
@mehmetdogru Do you have any logs that you can post?

mehmetdogru, 2020-02-28:
@EricBoiseLGSVL Here is the terminal output however .log files are empty.

$roslaunch op_global_planner op_global_planner.launch 
... logging to /home/leo-sim/.ros/log/fa0d6d1a-57b4-11ea-a2e9-3497f625a91c/roslaunch-leosim-pc-30362.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://leosim-pc:39885/

SUMMARY
========

PARAMETERS
 * /op_global_planner/enableDynamicMapUpdate: False
 * /op_global_planner/enableLaneChange: True
 * /op_global_planner/enableReplan: True
 * /op_global_planner/enableRvizInput: True
 * /op_global_planner/enableSmoothing: True
 * /op_global_planner/mapFileName: 
 * /op_global_planner/mapSource: 0
 * /op_global_planner/pathDensity: 0.75
 * /op_global_planner/velocitySource: 1
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    op_global_planner (op_global_planner/op_global_planner)

ROS_MASTER_URI=http://localhost:11311

process[op_global_planner-1]: started with pid [30377]
Received Lanes2800
Received Lines108
Received Points3065
Received dtLanes2800
Received StopLines12
Received Signals45
Received Vectors49
Received Nodes2800
Map Version 2
 >> Extracting Lanes ... 
[op_global_planner-1] process has died [pid 30377, exit code -11, cmd /home/leo-sim/Projects/autoware.ai/install/op_global_planner/lib/op_global_planner/op_global_planner __name:=op_global_planner __log:=/home/leo-sim/.ros/log/fa0d6d1a-57b4-11ea-a2e9-3497f625a91c/op_global_planner-1.log].
log file: /home/leo-sim/.ros/log/fa0d6d1a-57b4-11ea-a2e9-3497f625a91c/op_global_planner-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

---------------------------------------------------
_When i debug op_global_planner here is where the segmentation error occurs:_

**PlannerHNS::MappingHelpers::GetPointFromDataList MappingHelpers.cpp:2401
PlannerHNS::MappingHelpers::GetLanePoints MappingHelpers.cpp:2674
PlannerHNS::MappingHelpers::CreateLanes MappingHelpers.cpp:2544
PlannerHNS::MappingHelpers::ConstructRoadNetworkFromROSMessageV2 MappingHelpers.cpp:2303
GlobalPlanningNS::GlobalPlanner::MainLoop op_global_planner_core.cpp:435**
main op_global_planner.cpp:33
__libc_start_main 0x00007f02ddc66b97
_start 0x000055783182718a

--------------------

It is basically trying to read something that doesn't exist.
I tried to understand what is going on and as far as i get, op_global_planner doesn't accept "0" index for the node id and the centerlines(lanes) was looking kinda shifted. Also VectorMapBuilder by Tier4 has same approach that accepts "0" indexed id as **None** .So here is how "singleLaneRoad" Autoware Vector Map from lgsvl/autoware-data looks:

Editor: https://tools.tier4.jp/vector_map_builder/

[image]

--------------------------

Here is the vector map visualization for BorregasAve on Tier4 Vector Map Builder:

[image]

closed 2020-10-20 by EricBoiseLGSVL