lgsvl/simulator #641: Gps outputs data with offset

SeymurSm, 2020-02-20:
Hi Guys,

We are working with Gps sensor and Autoware to apply localization by means of Autoware. We noticed that the Gps sensor has returns longitude value with offset that is shown on below snapshot. I think both latitude and longitude should be values near to 0 if our map built near to Unity global origin. When I change 'IgnoreMapOrigin' flag to false on pre-built maps such as BorregasAvenue it also returns longitude value with offset which is problem for localization.

[image]

Furthermore, I have checked the code below that returns latitude and longitude values and not understand if there is problem with the conversion.

[image]

Thanks in advance.

hadiTab, 2020-02-26:
What versions of Autoware and the Simulator are you using?
GNSS localizer in Autoware only works in the Japanese coordinate system. There are also some issues related to working in coordinates near (0, 0).

SeymurSm, 2020-02-28:
@hadiTab  sorry for late reply. 
We are using simulator version 2019 by using ignore map origin flag as true in our GPS configuration.
Changin  `double x = easting - 500000d;` to ` double x = easting;`     in  MapOrigin.Conversion.cs has solved our problem. 

[image]

codezs09, 2020-03-02:
@SeymurSm I tried your solution and it works to correct GPS lat long. But it seems that the GPS_odometry sensor output also changes correspondingly. So if one is using the odometry information, it may not be a good idea to change that line.

closed 2020-10-20 by EricBoiseLGSVL