lgsvl/simulator #642: Map is shifted

MaxandreOgeret, 2020-02-21:
Hello,

I was using the version 2019.09 and the map BorregasAve to make my tests and I uodated to the last version 2020.01 and now the map is totally shifted, hundreds of meters away, and the gps localization is not aligned anymore.

ntutangyun, 2020-02-21:
@MaxandreOgeret Hi could you try to download the new maps from [lgsvl content map](https://content.lgsvlsimulator.com/maps/)

pedroexenberger, 2020-02-21:
@MaxandreOgeret may I ask you how do you get gps localization to work? I read autoware only accepts Japanese coordinates. NDT algorithm is fine, but sometimes the car gets lost while driving, so I think having gps along NDT would be better. Thank you!

shalinmehtalgsvl, 2020-02-26:
Hi @MaxandreOgeret ,

Changes had been made to maps so that they match real-world positions and orientations. We updated the HD maps with the latest release.

https://content.lgsvlsimulator.com/maps/borregasave/

closed 2020-04-08 by EricBoiseLGSVL