lgsvl/simulator #646: Lidar point_cloud message is not received (Apollo 5.0)

simon13579, 2020-02-22:
I am running Apollo 5.0 and Simulator 2019.12. In simulator's bridge view, there is a point_cloud message rolling, but in cyber_monitor  there is only camera related messages.

[image]

ntutangyun, 2020-02-23:
@simon13579 coud you try to press 'PageDown' button in `cyber_monitor` terminal. looks like some topics are in next page...

try to run `cyber_monitor` in a more larger terminal window.

simon13579, 2020-02-24:
@ntutangyun thanks, all the topics are shown when I run in a larger terminal window. But the problem is that the perception is turned off for unknown reason.  
[image]

I run cyber_launch start xx/xx/perception.launch to start the perception module, but only perception/motion_service is started.
[image]

Is there any method that I can find the root cause why perception module don't work?

ntutangyun, 2020-02-24:
@simon13579 I'm also new here. But somehow looks like the three submodules of perception are running.  How do you know that only `motion_service` is running and the other two crashed ?

If they do crash, does it happen before they receive any messages or after they processed some messages ? 

You could try to bring up the perception module for a while, before starting `bridge.sh`, see if they crash. 

If not, then you may launch the lgsvl and the `bridge.sh` to forward simulator messages. see if they crash or not .

simon13579, 2020-02-25:
Since the frame ratio of apollo/perception/obstacles and apollo/perception/traffic_light is zero, I guess they are crashed because they are not called at an interval.
My graphic card is GTX 1660ti whose memory is 6G, I am afraid that the performance of my graphic card might be an problem.

closed 2020-10-20 by EricBoiseLGSVL