lgsvl/simulator #647: Using Autoware with Python-api

NCTUyoung, 2020-02-23:
I am using **Autoware1.2** and **LGSVL 2020.1** on linux.
For some reason I need to control dynamic object/signal using python-api. 

I can successfully run Autoware with LGSVL or Python Api alone , but fail to connect these two things.
Is there any way to run LGSVL with Autoware and Pythonapi at the same time ?

The most similar tutorial in document is [https://www.lgsvlsimulator.com/docs/api-example-descriptions/](url) , which is running in api mode and connecting ros bridge in python api. 
However, I got no ros topic publishing even it is in connected status.
[image]

hadiTab, 2020-02-26:
@NCTUyoung are you actually using Autoware 1.2 or did you mean to say 1.12? Version 1.2 is very old and I don't think we ever supported it.

Have you looked at [this](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/22-connecting-bridge.py) quickstart example for the PythonAPI? I think it's a good example for what you are trying to achieve.

NCTUyoung, 2020-03-02:
@hadiTab  Sorry for the mistake, It should be 1.12. 
I  try this quickstart , but it does not work. 
I also check  the version PythonAPI(2020.01).
Finally, I downgrade the LGSVL version to **2019.10-rc1** , and it  work well.

Thank for your kindly reply

closed 2020-03-17 by NCTUyoung