lgsvl/simulator #649: How to find the ground truth for the point cloud?

daohu527, 2020-02-26:
1. You can obtain the ground truth through the "GroundTruth3DSensor", but its frequency may not be the same as the "LidarSensor", so you cannot accurately obtain the ground truth of the current point cloud.
2. You can use lidar save api to get the pcd file of the point cloud. But cannot get the groud truth of the 3d objects. 

How to solve this ? I think maybe should add the ground truth to the lidar save api?

EricBoiseLGSVL, 2020-02-28:
@daohu527 We are looking at ways to solve this, hopefully have a solution soon.

daohu527, 2020-02-28:
Thanks, look forward to this feature coming online!

cyberphysicalrobotics, 2020-02-29:
Two sensors frequency is set same. But ground truth 3D sensor is less frequency than 10 Hz. Lidar and ground truth 3D data is not synchronized each other. I'll share this with my team.

ntutangyun, 2022-01-22:
hi, any update on this feature ? thanks

EricBoiseLGSVL, 2022-01-24:
Sorry, with simulator being sunset, we will not be working on this feature request.

closed 2022-01-24 by EricBoiseLGSVL

daohu527, 2022-01-25:
what does "simulator is sunset" mean ? Does it mean that it will not be developed in the future?

EricBoiseLGSVL, 2022-01-25:
yes, LG is no longer supporting simulator.  Please see [here](https://www.svlsimulator.com/news/2022-01-20-svl-simulator-sunset).

daohu527, 2022-01-25:
This is sad news, open source brings benefits but leaves many questions, thanks for your contribution : )

I'm one of the first few people to follow this project, and it's a pity that this project is no longer maintained. This has also strengthened my determination to do something that requires feelings

EricBoiseLGSVL, 2022-01-25:
Thanks for working with us for so long.  Yes, it is sad but projects move on and new opportunities arise.  Good luck to you and thanks for all the help here in the issues page :)