lgsvl/simulator #651: Lidar Localization in Apollo5.0 and LGSVL simulator

JAckLosingHeart, 2020-02-26:
Hello,

Recently I tried to run Apollo5.0 with the simulator using MSF Localization. However, I always got a warning in Dreamview which's "Warning: Lidar Localization Is Abnormal." The messages in the Localization log are like "PcdProcess: Discard a lidar msg because can't get predict pose. " and "Receive a lidar msg, but can't query predict pose." The Lidar Localization didn't seem to work. Then I read the code in localization_lidar_process.cc and found it's something wrong in **pose_forecastor_->GetBestForcastPose(lidar_time, -1, init_pose, &forecast_pose)** , the code of which is close-source. So I got stuck with getting this work.
The following is what I did to generate the local_map:

1. First, I run Apollo5.0 in the simulator and used cyber_recorder to record the Point Cloud Data and pose files.
2. Then I used cyber_record_parser and poses_interpolator to process the record files.
3. Finally, I used lossless_map_creator and lossless_map_to_lossy_map to generate the local_map for Shalun.

My question is:
The apollo5.0 uses RTK Localization by default, and I saw there aren't local maps in any map provided. Does simulator support MSF Localization in Apollo5.0? If it does, how can I get Apollo5.0 work with MSF Localization, and would you like to share some experiences with me? Thank you so much.

EricBoiseLGSVL, 2020-02-27:
@hadiTab Do you know what is failing here?

EricBoiseLGSVL, 2020-10-20:
MSF localization is supported in Apollo 5.0 but there are issues with latest Apollo.  We recommend using RTK for now.

closed 2020-10-20 by EricBoiseLGSVL

mengen100, 2021-04-16:
@EricBoiseLGSVL Also want to check are the above steps by @JAckLosingHeart correct for creating a local_map to run the MSF localization module? I am having trouble in getting data in the channel **msf_lidar** too.

EricBoiseLGSVL, 2021-04-16:
Sorry, we are using RTK, not sure on issues with MSF.  Any idea @hadiTab ?