lgsvl/simulator #652: I can'y control the car in Lgsvl with Logitech G920

ozguradas, 2020-02-27:
Hello all,
I am currently using the lgsvl 2020.01 version. Before we add our map and car, we tried to control standard map (Single road or san fran.) and standard car (Jaguar with no bridge and bridge autoware). Anyway i'm new on this. Is there anything that i have to do before plug in the g920.

I can control the car with keyboard.
I downloaded and installed Logitech G Hub. (Logitech Gaming Software doesn't support my system. I have Dell M15).

So what can i do now ?

Thanks for answers.

EricBoiseLGSVL, 2020-02-28:
@ozguradas we have issues with Unity recognizing the Logitech drivers sometimes,  I usually just unplug and re-plug the usb ports.  You can check that Unity is reading the drivers correctly with the Input system debug panel.  Window -> Analysis -> Input Debugger
Also, be sure to add a [WheelControlSensor](https://www.lgsvlsimulator.com/docs/sensor-json-options/#wheel-control).

ozguradas, 2020-02-28:
Well, thanks for the answer. I didn't know the input debugger section. Now i can control the car via g920.

closed 2020-02-28 by ozguradas