lgsvl/simulator #656: LGSVL launch issue

TNAV1, 2020-02-28:
Last year I used LGSVL 2019.05 with Autoware 1.12.0.  I have since deleted and would like to start again.  I see that 2019.05 is no longer available so I am using 2020.01.  
1) When I launch the .exe file it allows me to pick the graphics options.  When I click done it closes the window and nothing happens.
2) When I load the source code into Unity 2019.1.10f1 and run it produces the following errors.
System.Exception: Cannot load. Incorrect path: Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipelineResources/Texture/DebugFont.tga Null returned.
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.Load (System.String path, System.Type type, System.Boolean builtin) [0x00036] 

Any suggestions?  I would prefer to get option 2 working.

EricBoiseLGSVL, 2020-02-29:
@TNAV1 
1) You may need to update drivers.  Can you post driver details?
2) Are you cloning and do you have git lfs installed?

luodewei, 2020-02-29:
I have the same 2. problem, and I am run Unity 2019.1.10f1 on Windows10 not ubuntu

TNAV1, 2020-03-01:
> @TNAV1
> 
>     1. You may need to update drivers.  Can you post driver details?
> 
>     2. Are you cloning and do you have git lfs installed?

git config --get-regex lfs
filter.lfs.clean git-lfs clean -- %f
filter.lfs.smudge git-lfs smudge -- %f
filter.lfs.process git-lfs filter-process
filter.lfs.required true

+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.59       Driver Version: 440.59       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 106...  Off  | 00000000:09:00.0  On |                  N/A |
|  0%   52C    P0    25W / 120W |    258MiB /  6075MiB |      3%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1070      G   /usr/lib/xorg/Xorg                            18MiB |
|    0      1239      G   /usr/bin/gnome-shell                          48MiB |
|    0      1572      G   /usr/lib/xorg/Xorg                           102MiB |
|    0      1704      G   /usr/bin/gnome-shell                          83MiB |
|    0      3254      G   /usr/lib/firefox/firefox                       1MiB |
+-----------------------------------------------------------------------------+

EricBoiseLGSVL, 2020-03-02:
@TNAV1 @luodewei Please post Playerlogs for editor and player. Drivers look fine for @TNAV1 Windows should not have this issue.

TNAV1, 2020-03-04:
@EricBoiseLGSVL I am running Ubuntu 18.04, ROS Melodic, Autoware 1.12.0, Unity 20919.1.10f.1, LGSVL 2020.01.  Does that give any additional insight?

EricBoiseLGSVL, 2020-03-04:
@luodewei @TNAV1 we think we have seen this bug with ROS and autoware.  We ended up hard booting the pc and it was fixed but we are still investigating.  For Windows we are also trying to update to see if this is causing the issue.  I think it is the ROS bridge on simulator side.  Please post any other details as you debug and we will do the same.

TNAV1, 2020-03-04:
@EricBoiseLGSVL I have rebooted with the same results.  Not sure what you mean for the playerlogs.  A disc search did not turn up anything.  Could you tell me where to look for them in Ubuntu?

EricBoiseLGSVL, 2020-03-04:
@TNAV1 logs are [here](https://docs.unity3d.com/Manual/LogFiles.html)
When you cloned did you have git lfs installed.  It looks like you are missing files.  Also can you screen shot your packages installed, Window -> Package Manager

TNAV1, 2020-03-04:
@EricBoiseLGSVL 
[Player.log](https://github.com/lgsvl/simulator/files/4284599/Player.log)
[Editor.log](https://github.com/lgsvl/simulator/files/4284600/Editor.log)
[upm.log](https://github.com/lgsvl/simulator/files/4284601/upm.log)

TNAV1, 2020-03-04:
[image]
[image]
[image]

I believe git lfs was installed - see output from 3 days ago 
https://github.com/lgsvl/simulator/issues/656#issuecomment-593046149

EricBoiseLGSVL, 2020-03-04:
@TNAV1 can you post packages in project, not all please.  It's a toggle top left

EricBoiseLGSVL, 2020-03-04:
@TNAV1 what map are you trying to load? What is your GPU mem, 6gb?  If You have a large map you maybe running out of memory.

TNAV1, 2020-03-04:
[image]

Sorry, I misunderstood.

TNAV1, 2020-03-04:
I don't even get to the option to select a map.

TNAV1, 2020-03-04:
@EricBoiseLGSVL, I am probably doing something wrong on a very basic level.  See video
https://drive.google.com/file/d/1j8rpQr2XYvMt7VAEzkyzDOLrX8BCaY0f/view?usp=sharing

martins-mozeiko, 2020-03-04:
Your Unity uses OpenGL for rendering (you can see in title "OpenGL 4.5" string). HDRP rendering in Unity on Linux requires to use Vulkan on Linux. Make sure you have `libvulkan1` package installed on Ubuntu. You can verify that Vulkan works correctly by running `vulkaninfo` in commandline.

ck-leo, 2020-03-04:
[Player.log](https://github.com/lgsvl/simulator/files/4285313/Player.log)
@EricBoiseLGSVL I have encountered the problem that I cannot open browser when I click the button "open browser ..." in the LGSVL simulator window.
[image]
My computer configuration is follow:
Windows 10
i7 9750H @2.60GHz
16G DDR4
GTX 2060 6G
Is there some suggestions for this problem? Thanks.

EricBoiseLGSVL, 2020-03-04:
@martins-mozeiko Yes, thanks for finding that.  Linux requires vulkan.  @TNAV1 Also, I see your materials and textures are blank.  This is because you have not unpacked the git lfs files.  Please make sure git lfs is installed and then git pull.  Please be sure you are cloning the repo, not downloading a zip of the project.  It needs to be cloned so git lfs can even work.  Let me know if this helps

@ck-leo please check the web browser.  Then check if you can navigate to localhost:8080 in the browser.  Lastly, please post the player.log to check for errors and what you are on Linux/Windows/ReleaseBuild/Editor/CreatedBuild.  This would be a good new issue for others to be able to search

TNAV1, 2020-03-04:
@martins-mozeiko - vulkan was the issue thanks.

TNAV1, 2020-03-05:
@EricBoiseLGSVL - I cloned the repo and git lfs is installed to the best of my knowledge.  It seems to have gotten further.  However, I was expecting to see the "choose map" screen when I launched the player.
https://drive.google.com/file/d/1cu3aw0q6hln6XQZEp-NaqKGrKzwgXo4j/view?usp=sharing

EricBoiseLGSVL, 2020-03-05:
@TNAV1 that video didn't record the screen, just the one window

TNAV1, 2020-03-05:
@EricBoiseLGSVL sorry about that...
https://drive.google.com/file/d/1TvKIB3eeYPSJxP0xC0kK2dmF0k8yQwyQ/view?usp=sharing

TNAV1, 2020-03-05:
The executable works, but still need to get the source code working in Unity.
[image]

TNAV1, 2020-03-05:
@EricBoiseLGSVL for the Unity approach I only cloned.  Did I need to make or build the files.  (I am new to Ubuntu).

EricBoiseLGSVL, 2020-03-05:
@TNAV1 in editor you need to open Simulator -> Developer Settings and select the vehicle you have setup in the WebUI.  It reads the vehicles saved in database file in persistent data folder and loads that bundle.  If you load editor from LoaderScene.scene, just load the webui and bundles will load from that.  If you create new vehicles and maps, be sure to build with Simulator -> Build... and then use webui to point at the new bundles.  Note, be sure to add sensor parameters to new vehicles, e.g., keyboard control sensor and other sensors you need.  Look at preconfigured provided as examples.

TNAV1, 2020-03-06:
@EricBoiseLGSVL is there documentation that you can point me to to use LGSVL in Unity?  My 1st step is I want to run SanFrancisco and the Jag (Autoware) and make minor tweaks.

TNAV1, 2020-03-06:
@EricBoiseLGSVL I found the LoaderUI->StartButton in Unity.  Pressing it it tries to open a browser window, flashes content then goes blank.  (similar to when I did not have Vulkan installed.  Any ideas?
https://drive.google.com/file/d/1Z9s8OKAWw0Q1eX5rFpgBEcm-YBu3KsDI/view?usp=sharing

EricBoiseLGSVL, 2020-03-06:
@TNAV1 Try Ctrl-R to refresh the page when focused on the browser

TNAV1, 2020-03-06:
@EricBoiseLGSVL no luck

EricBoiseLGSVL, 2020-03-06:
Can you post here the url that the webui is trying to reach? I can't see it from video

TNAV1, 2020-03-06:
http://localhost:8080/

TNAV1, 2020-03-08:
I found this and am investigating... https://github.com/lgsvl/simulator/issues/305

closed 2020-03-08 by TNAV1