lgsvl/simulator #657: GPS data error

chronofei, 2020-02-28:
The GPS data in simulator is different from the one in the nmea_sentence msgs? I don't know where is wrong.

EricBoiseLGSVL, 2020-02-29:
@hadiTab Do you know what could be the issue?

EricBoiseLGSVL, 2020-04-17:
@chronofei we have fixes for the nmea in the latest release.  Are you still having this issue

closed 2020-10-20 by EricBoiseLGSVL

JM998, 2021-02-07:
I have the same problem with the latest version, how to deal with it? 
the version is 2020.06 for windows, is this problem fixed in the latest version of windows?

JM998, 2021-02-07:
@EricBoiseLGSVL  I have the same problem with the latest version, how to deal with it?
the version is 2020.06 for windows, is this problem fixed in the latest version of windows?

EricBoiseLGSVL, 2021-02-08:
Please look at this issue, we are currently investigating.  #1077

JM998, 2021-02-13:
I don't know