lgsvl/simulator #658: Unable to get consistent connection through ROS bridge - screen flickers

pranavbajoria93, 2020-02-29:
I am trying to setup LGSVL with ROS on a ubuntu 18.04 with NVIDIA Titan X graphics card with all the latest drivers installed fine.

Without the ROS bridge, the simulator runs smoothly. But upon connection with ROS bridge, the screen starts flickering and the connection status continuously fluctuates from connected to disconnected at both the simulator's end and the rosbridge's end.

[Ros_bridge_failed connection.zip](https://github.com/lgsvl/simulator/files/4269545/Ros_bridge_failed.connection.zip)

First, I tried using the standard rosbridge suite with a sudo install as per the official guide provided by ROS. The connection was unstable and there were no rostopics listed. I am wondering why that could be the case. 

Then, I pulled the LGSVL's rosbridge_suite into a catkin_ws and did a catkin make and then used the new rosbridge_server. This time the connection was still unstable but I could list the ros topics. However, none of the topics have any messages published. 

Any help would be appreciated.

Earlier from the issue From the player log, it seems that the Vulkan drivers are installed fine.

Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce GTX TITAN X] id=[17c2]

Here is a reference to my comment on a similar issue. [https://github.com/lgsvl/simulator/issues/445#issuecomment-592247876](url)

pranavbajoria93, 2020-02-29:
Player log:

```
Desktop is 2560 x 1440 @ 60 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: count=79
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[... 941 lines cut ...]
Unloading 0 Unused Serialized files (Serialized files now loaded: 0)

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 5507.
Total: 7.405760 ms (FindLiveObjects: 0.268120 ms CreateObjectMapping: 0.121943 ms MarkObjects: 7.000122 ms  DeleteObjects: 0.009079 ms)

Unloading 0 Unused Serialized files (Serialized files now loaded: 0)

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 5507.
Total: 8.148662 ms (FindLiveObjects: 0.263581 ms CreateObjectMapping: 0.124177 ms MarkObjects: 7.742465 ms  DeleteObjects: 0.012013 ms)

Unloading 0 Unused Serialized files (Serialized files now loaded: 0)

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 5507.
Total: 11.240536 ms (FindLiveObjects: 0.378539 ms CreateObjectMapping: 0.228940 ms MarkObjects: 10.611056 ms  DeleteObjects: 0.013549 ms)

Unloading 0 Unused Serialized files (Serialized files now loaded: 0)

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 5507.
Total: 7.128909 ms (FindLiveObjects: 0.260369 ms CreateObjectMapping: 0.114679 ms MarkObjects: 6.736261 ms  DeleteObjects: 0.010895 ms)

Unloading 0 Unused Serialized files (Serialized files now loaded: 0)

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 5507.
Total: 9.565532 ms (FindLiveObjects: 0.300388 ms CreateObjectMapping: 0.123689 ms MarkObjects: 9.122809 ms  DeleteObjects: 0.012362 ms)
```

pranavbajoria93, 2020-02-29:
An excerpt from my player log:

`2/28/2020 5:33:03 PM|Fatal|WebSocket.connect|Connection refused
2/28/2020 5:33:03 PM|Fatal|WebSocket.connect|Connection refused
2/28/2020 5:33:03 PM|Fatal|WebSocket.connect|Connection refused
2/28/2020 5:33:03 PM|Fatal|WebSocket.connect|Connection refused
2/28/2020 5:33:03 PM|Fatal|WebSocket.connect|Connection refused
2/28/2020 5:33:05 PM|Error|WebSocket.sendBytes|Unable to write data to the transport connection: The socket has been shut down.
A send has been interrupted.
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

2/28/2020 5:33:05 PM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2|System.IO.IOException: Unable to read data from the transport connection: Connection reset by peer. ---> System.Net.Sockets.SocketException: Connection reset by peer
                             at System.Net.Sockets.Socket.EndReceive (System.IAsyncResult asyncResult) [0x00012] in <0079a30f96a047348857e1cecc6c638a>:0 
                             at System.Net.Sockets.NetworkStream.EndRead (System.IAsyncResult asyncResult) [0x00057] in <0079a30f96a047348857e1cecc6c638a>:0 
                              --- End of inner exception stack trace ---
                             at System.Net.Sockets.NetworkStream.EndRead (System.IAsyncResult asyncResult) [0x0009b] in <0079a30f96a047348857e1cecc6c638a>:0 
                             at WebSocketSharp.Ext+<>c__DisplayClass56_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x00002] in <2e9e916731394d0aa2e65b2f47be4906>:0 
Failed to read GPU texture
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Failed to read GPU texture
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

2/28/2020 5:33:06 PM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2|System.IO.IOException: Unable to read data from the transport connection: Connection reset by peer. ---> System.Net.Sockets.SocketException: Connection reset by peer
                             at System.Net.Sockets.Socket.EndReceive (System.IAsyncResult asyncResult) [0x00012] in <0079a30f96a047348857e1cecc6c638a>:0 
                             at System.Net.Sockets.NetworkStream.EndRead (System.IAsyncResult asyncResult) [0x00057] in <0079a30f96a047348857e1cecc6c638a>:0 
                              --- End of inner exception stack trace ---
                             at System.Net.Sockets.NetworkStream.EndRead (System.IAsyncResult asyncResult) [0x0009b] in <0079a30f96a047348857e1cecc6c638a>:0 
                             at WebSocketSharp.Ext+<>c__DisplayClass56_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x00002] in <2e9e916731394d0aa2e65b2f47be4906>:0 
Failed to read GPU texture
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Failed to read GPU texture
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Failed to read GPU texture
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

2/28/2020 5:33:07 PM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2|WebSocketSharp.WebSocketException: The header of a frame cannot be read from the stream.
                             at WebSocketSharp.WebSocketFrame.processHeader (System.Byte[] header) [0x0001a] in <2e9e916731394d0aa2e65b2f47be4906>:0 
                             at WebSocketSharp.WebSocketFrame+<>c__DisplayClass73_0.<readHeaderAsync>b__0 (System.Byte[] bytes) [0x00000] in <2e9e916731394d0aa2e65b2f47be4906>:0 
                             at WebSocketSharp.Ext+<>c__DisplayClass56_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x0009e] in <2e9e916731394d0aa2e65b2f47be4906>:0 
2/28/2020 5:33:08 PM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2|WebSocketSharp.WebSocketException: The header of a frame cannot be read from the stream.
                             at WebSocketSharp.WebSocketFrame.processHeader (System.Byte[] header) [0x0001a] in <2e9e916731394d0aa2e65b2f47be4906>:0 
                             at WebSocketSharp.WebSocketFrame+<>c__DisplayClass73_0.<readHeaderAsync>b__0 (System.Byte[] bytes) [0x00000] in <2e9e916731394d0aa2e65b2f47be4906>:0 
                             at WebSocketSharp.Ext+<>c__DisplayClass56_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x0009e] in <2e9e916731394d0aa2e65b2f47be4906>:0 
2/28/2020 5:33:09 PM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2|System.IO.IOException: Unable to read data from the transport connection: Connection reset by peer. ---> System.Net.Sockets.SocketException: Connection reset by peer
                             at System.Net.Sockets.Socket.EndReceive (System.IAsyncResult asyncResult) [0x00012] in <0079a30f96a047348857e1cecc6c638a>:0 
                             at System.Net.Sockets.NetworkStream.EndRead (System.IAsyncResult asyncResult) [0x00057] in <0079a30f96a047348857e1cecc6c638a>:0 
                              --- End of inner exception stack trace ---
                             at System.Net.Sockets.NetworkStream.EndRead (System.IAsyncResult asyncResult) [0x0009b] in <0079a30f96a047348857e1cecc6c638a>:0 
                             at WebSocketSharp.Ext+<>c__DisplayClass56_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x00002] in <2e9e916731394d0aa2e65b2f47be4906>:0 
2/28/2020 5:33:10 PM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2|WebSocketSharp.WebSocketException: The header of a frame cannot be read from the stream.
[... cut ...]
                             at WebSocketSharp.WebSocketFrame+<>c__DisplayClass73_0.<readHeaderAsync>b__0 (System.Byte[] bytes) [0x00000] in <2e9e916731394d0aa2e65b2f47be4906>:0 
                             at WebSocketSharp.Ext+<>c__DisplayClass56_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x0009e] in <2e9e916731394d0aa2e65b2f47be4906>:0 
Stopping simulation with id 6
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Unloading 2 Unused Serialized files (Serialized files now loaded: 2)
UnloadTime: 21.629426 ms
Compute dispatch: missing UAV ID %d
 
(Filename:  Line: 0)

Compute dispatch: missing texture ID %d
 `

EricBoiseLGSVL, 2020-03-04:
@hadiTab do you know what might be the issue here?

pranavbajoria93, 2020-03-06:
Updates to the issue:
I tried 4 different configurations with LGSVL running on Ubuntu and alternatively on Windows.
Here's what happened.

- Simulator on Ubuntu, ROSbridge websocket (LGSVL's fork) on Ubuntu:  The connection keeps alternating between connected and disconnected. The Ros topics can not echo any data even though the topics are listed.

- Simulator on Ubuntu, Rosbridge via the sensor.launch in the Autoware Docker: Rosbridge connection is stable and the messages and images can be seen in the RVIZ. However the LGSVL simulator continues to flicker because of the GPU texture issue

- Simulator on windows, ROSbridge websocket (LGSVL's fork) on Ubuntu: connection unstable and screen flickers, no rostopic can be echoed.

- Simulator on windows, Rosbridge via the sensor.launch in the Autoware Docker: stable connection with no screen flicker.

From this I understood that there is surely something wrong with my rosbridge installation as both Windows and Ubuntu simulator fail to gain a consistent connection. Secondly, the simulator on ubuntu flickers regardless of consistent connection with autoware's rosbridge. 

@hadiTab @EricBoiseLGSVL Any insights on what might be going wrong here?

EricBoiseLGSVL, 2020-04-08:
@pranavbajoria93 We have a new release updating Unity going out in the next few days.  This fixes many Vulkan issues.  We are also refactoring bridge code and hope to have a better solution soon

pranavbajoria93, 2020-04-13:
Update with the new 2020.03 release:

1.  On Ubuntu, the simulator camera doesn't flicker to the TL camera anymore and the render is constantly fixed at the follow camera. So main issue is resolved.
2. The simulator has consistent connection with the stock ROSbridge websocket and all topics can are listed and messages are being sent. However, the LGSVL's rosbridge suite continues to give the same problem.

Closing the issue.

closed 2020-04-13 by pranavbajoria93

EricBoiseLGSVL, 2020-04-14:
@pranavbajoria93 Glad the update fixed Vulkan issues.  We are working on improving the bridge and hope to have a better solution soon.  Thanks for your patience.