lgsvl/simulator #659: Error message while running lgsvl : ''Invalid: Out of date Vehicle AssetBundle''

Avps1, 2020-02-29:
Hi, 
I got this message : `Invalid: Out of date Vehicle AssetBundle. Please check content website for updated bundle or rebuild the bundle.` while trying to run the lgsvl simulation (See attached screenshot). So I reinstalled lgsvl (linux 2020.01 version) from the latest repo : [(https://github.com/lgsvl/simulator/releases/latest)] and ran it again, but with no luck. The same error message appears. I thought my lgsvl was outdated, but as I said, the error persisted despite reinstalling lgsvl . 

System specs : Ubuntu 18.04 
Nvidia-Driver Version: 440.59       CUDA Version: 10.2 
GTX 1080 GRaphics Card. 
Please help. LGSVL was running perfectly in my machine even 2 days ago. 
[image]

ntutangyun, 2020-03-01:
please download the latest vehicle assetBundle from [here](https://content.lgsvlsimulator.com/vehicles/)

Avps1, 2020-03-07:
Many thanks for your response.

closed 2020-03-11 by Avps1