lgsvl/simulator #660: Traffic light can't be recognized and large offset on trajectory

kanbei, 2020-03-01:
Hello!
I ran Apollo5.0 with Lgsvl, and met some problems.
First, the perception module could be enabled but traffic light couldn't. Once the car run to the intersection, it always stop, although the traffic light become green.
Second, when automatically driving, the car could hardly run along the path, it turned left and right, seemed that large offset on the trajectory.
Finally, if I want to know the coordinate values of the given trajectory by Apollo and the real trajectory by the car, what kind of log records these coordinate values?
Thank you.

[image]
----------------------------------------------------
This is the planning log. 

Log file created at: 2020/03/01 19:43:13
Running on machine: in_dev_docker
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
I0301 19:43:13.736858  3431 class_loader_utility.h:78]  registerclass:PlanningComponent,apollo::cyber::ComponentBase,/apollo/bazel-bin/modules/planning/libplanning_component.so
E0301 19:43:13.819878  3488 reference_line_provider.cc:193]  Routing is not ready.
E0301 19:43:13.870038  3495 reference_line_provider.cc:193]  Routing is not ready.
E0301 19:43:13.920229  3486 reference_line_provider.cc:193]  Routing is not ready.
E0301 19:43:13.970335  3489 reference_line_provider.cc:193]  Routing is not ready.
E0301 19:43:14.020470  3485 reference_line_provider.cc:193]  Routing is not ready.
E0301 19:43:14.070624  3497 reference_line_provider.cc:193]  Routing is not ready.
E0301 19:43:14.120776  3487 reference_line_provider.cc:193]  Routing is not ready.
E0301 19:43:14.170919  3486 reference_line_provider.cc:193]  Routing is not ready.
E0301 19:43:14.221063  3485 reference_line_provider.cc:193]  Routing is not ready.
E0301 19:43:14.271195  3484 reference_line_provider.cc:193]  Routing is not ready.
I0301 19:43:14.279661  3499 planning_component.cc:57]  Received routing data: run routing callback.timestamp_sec: 1583059390.4533658
module_name: "routing"
sequence_num: 0
E0301 19:43:14.321336  3484 reference_line_provider.cc:193]  Routing is not ready.

......

W0301 19:43:33.029922  3484 reference_line_provider.cc:253]  Reference line is NOT ready.
E0301 19:43:33.029948  3484 reference_line_provider.cc:255]  Failed to use reference line latest history
E0301 19:43:33.030041  3484 on_lane_planning.cc:266]  PLANNING_ERROR: Failed to create reference line
E0301 19:43:33.245293  3493 piecewise_jerk_problem.cc:104]  failed optimization status:	primal infeasible
E0301 19:43:33.245954  3493 piecewise_jerk_path_optimizer.cc:192]  piecewise jerk path optimizer failed
E0301 19:43:33.282368  3493 piecewise_jerk_problem.cc:104]  failed optimization status:	primal infeasible
E0301 19:43:33.282757  3493 piecewise_jerk_path_optimizer.cc:192]  piecewise jerk path optimizer failed
E0301 19:43:33.282999  3493 path_optimizer.cc:41]  Reference Line 0_0 is not drivable after PiecewiseJerkPathOptimizer
E0301 19:43:33.283200  3493 lane_follow_stage.cc:165]  Failed to run tasks[PiecewiseJerkPathOptimizer], Error message: Path Optimizer failed to generate path
E0301 19:43:33.283470  3493 lane_follow_stage.cc:254]  Path fallback due to algorithm failure
E0301 19:43:33.285091  3493 path_data.cc:40]  Should NOT set discretized path when reference line is nullptr. Please set reference line first.
E0301 19:43:33.285389  3493 lane_follow_stage.cc:277]  Speed fallback due to algorithm failure
E0301 19:43:33.285581  3493 speed_profile_generator.cc:43]  Fallback using piecewise jerk speed!
W0301 19:43:33.285768  3493 speed_profile_generator.cc:46]  init_v = 4.3189e-05, init_a = -0.260451
E0301 19:43:33.344827  3493 piecewise_jerk_problem.cc:104]  failed optimization status:	maximum iterations reached
E0301 19:43:33.345185  3493 speed_profile_generator.cc:88]  Piecewise jerk fallback speed optimizer failed!
E0301 19:43:33.345409  3493 speed_profile_generator.cc:129]  Slowing down the car within a constant deceleration with fallback stopping profile.
E0301 19:43:33.345682  3493 reference_line_info.cc:500]  path data is empty
E0301 19:43:33.345861  3493 lane_follow_stage.cc:192]  Fail to aggregate planning trajectory.
E0301 19:43:33.346055  3493 scenario.cc:79]  Stage 'LANE_FOLLOW_DEFAULT_STAGE' returns error
E0301 19:43:33.346256  3493 on_lane_planning.cc:489]  planner failed to make a driving plan
E0301 19:43:33.346463  3493 on_lane_planning.cc:324]  Planning failed:PLANNING_ERROR: planner failed to make a driving plan
E0301 19:43:33.396245  3493 piecewise_jerk_problem.cc:104]  failed optimization status:	primal infeasible
E0301 19:43:33.396672  3493 piecewise_jerk_path_optimizer.cc:192]  piecewise jerk path optimizer failed
E0301 19:43:33.428977  3493 piecewise_jerk_problem.cc:104]  failed optimization status:	primal infeasible
E0301 19:43:33.429072  3493 piecewise_jerk_path_optimizer.cc:192]  piecewise jerk path optimizer failed
E0301 19:43:33.429131  3493 path_optimizer.cc:41]  Reference Line 0_0 is not drivable after PiecewiseJerkPathOptimizer
E0301 19:43:33.429160  3493 lane_follow_stage.cc:165]  Failed to run tasks[PiecewiseJerkPathOptimizer], Error message: Path Optimizer failed to generate path
......

luqiang21, 2020-03-02:
@kanbei Have you checked your fps on simulator? If it is less than 30, that means you need a better GPU. It happens also on our machine if the fps is about 10. The traffic light module failure could also due to the same reason.

ntutangyun, 2020-03-02:
@kanbei besides a better GPU, in my case, your CPU and memory might also constrain the fps, as it involves huge amount of data transferred between LGSVl simulator and Apollo stack.

EricBoiseLGSVL, 2020-05-08:
@kanbei Was the issue low fps?

closed 2020-07-31 by lemketron