lgsvl/simulator #661: Can‘t open Jaguar2015XE model normally, all colors are pink

luodewei, 2020-03-02:
Hello,
I downloaded  Jaguar2015XE https://content.lgsvlsimulator.com/vehicles/jaguarxe2015/,
and open it in unity editor(windows10), why the car colors are all pink?
seem most shader all lost. 
[image]

EricBoiseLGSVL, 2020-03-02:
@luodewei  this is an indicator that you are missing the textures or you are using the non HDRP shaders.  Be sure when you clone the Jaguar repo, you have git lfs installed.

luodewei, 2020-03-04:
> @luodewei this is an indicator that you are missing the textures or you are using the non HDRP shaders. Be sure when you clone the Jaguar repo, you have git lfs installed.

Yes, I have git lfs installed, and clone the Jaguar repo under Ubuntu and open it in windows10. The size of repo I clone is 31.8MB, I think it is correct

EricBoiseLGSVL, 2020-03-04:
@luodewei Are the shaders on the materials for the vehicle HDRP/Lit?  Have the shaders compiled?  Please go to Edit -> Render Pipeline -> Update Project materials to High Definition materials

luodewei, 2020-03-05:
Finally I open the vehicle model normally, thanks @EricBoiseLGSVL 
follows are the steps:
1.Create a HDRP project, then create a scene
2.Assets-Create-Rendering-HDRP Asset
3.Edit-Project Settings-Graphics-Scriptable Render Pipeline Settings, Drop 2. HDRP Asset to here
4.Copy the Jaguar2015XE repo to project Assets folder
5.Open Jaguar2015XE.prefab
6.Edit -> Render Pipeline -> Update Project materials to High Definition materials

closed 2020-03-05 by luodewei