lgsvl/simulator #662: Define Apollo route programmatically

vatna, 2020-03-02:
I am using Apollo 5.0 with the Python API and would like to be able to define the route in Apollo programmatically instead of having to do this manually. This was requested before in issue #449 and I was wondering if there was any further news on this? Also, #449 mentioned that LG do this with some Python code over websocket. Would you be able to publish the code to do this?

EricBoiseLGSVL, 2020-03-02:
@vatna We are looking into solutions for this level of control but we do not have it in our current roadmap.

vatna, 2020-03-02:
Thanks for the update @EricBoiseLGSVL. Would you be able to publish the code mentioned in #449, or give some tips as to how to do it?

closed 2020-03-12 by vatna

EricBoiseLGSVL, 2020-03-13:
@vatna here is our code that we use to set EGO from python api
[dreamview-cyber-master.zip](https://github.com/lgsvl/simulator/files/4330725/dreamview-cyber-master.zip)

vatna, 2020-03-17:
Thanks @EricBoiseLGSVL ! That also contains some other code that will be extremely useful for what we're doing.

fabriquant, 2020-04-22:
Hello @EricBoiseLGSVL ,

I am looking at doing the same thing but with Autoware.AI.
Would you have some pointers on how to get started ?
For example, should I publish some ROS messages to do it and would I be constrained by the Autoware GUI (so much is done in the dialog box)?

Many thanks

EricBoiseLGSVL, 2020-04-22:
@fabriquant You will need to publish a ROS message.  We have not done this yet sorry