lgsvl/simulator #663: Logging ground truth data from LG simulator

vatna, 2020-03-02:
I am using Apollo 5.0 with the Python API and would like to log the ground truth data of all vehicles from the LG simulator. Is there a way to do this? A similar question was asked in issue #493 and it was suggested that we log Apollo data. The problem with this is that if Apollo doesn't see the other NPC vehicles, for example it is too far away, there will be no data. So the best way would be to log this data from the LG simulator, and we would then have the ground truth data.

The type of data we'd want for all the vehicles, ego and all NPC, is:
- Vehicle positions. As x,y,z,orientation but also relative to the lane that vehicle is following which could be of form: lane id, distance along that lane
- Vehicle speeds & accelerations, probably using vehicle's reference frame
- Vehicle control status (accelerator, brake, steering, handbrake, etc.). This would probably be simplified for NPC vehicles
- Collision status
- Simulation time

EricBoiseLGSVL, 2020-03-02:
@vatna I would suggest creating a new sensor plugin that gets all data from NPC and pedestrian managers.  You would need to write some code for collision status but it wouldn't be much.

vatna, 2020-03-02:
I've had a look at the sensor plugin documentation which seems to create a new camera. When you mention getting "all the data from NPC and pedestrian managers", would you be able to provide more details on how to do it, or point me to the relevant documentation? I haven't used Unity before unfortunately, so I have quite a bit to learn in order to do this.

EricBoiseLGSVL, 2020-03-03:
@vatna No worries, I'll try to break up steps.  Glad you are using Unity and our Simulator.
1) Since you are new to Unity, let's just make a new sensor, not a plugin.
2) In an empty scene create a new GameObject in the Hierarchy panel, name it GroundTruthSceneSensor and zero out it's position and rotation and set Scale to 1,1,1.
3) Now let's just duplicate GroundTruth3DSensor.cs because it will be very similar to what you want to do.
4) Assets -> Scripts -> Sensors -> GroundTruth3DSensor, select and press Ctrl-D to dupe.
5) Compiler will freak out, no worries.  Open the duped class and rename it GroundTruthSceneSensor in the code.
6) Rename JSONGroundTruth3D class inside the namespace to JSONGroundTruthScene
7) In Editor, select class, hit F2, and rename dupe to GroundTruthSceneSensor.cs
8) Now Unity compiler will calm down and you can add the class to the new game object in the hierarchy panel.
9) Then drag the game object with the new class on it to Assets -> Prefabs -> Sensors folder in the Project panel.  You created a prefab, woot.
10) Delete the prefab in the hierarchy panel and now only select and edit the new prefab in the Project panel.  Prefab objects in the Hierarchy are instances of the prefabs, if not overwritten, will never keep changes.
11) Go to  Assets -> Resources -> RuntimeSetting.asset.
12) Increase the SensorPrefabs array by one and drag the new sensor into the list
13) Boom, now you have a new sensor.  This is a copy of the GroundTruth3DSensor but can be changed to fit your needs.
14) In GroundTruthSceneSensor.cs code get references to SimulatorManager.Instance.AgentManager, SimulatorManager.Instance.NPCManager and Simulator.Instance.PedestrianManager
15) Now all these managers have references to the CurrentAgent/NPCs/Pedestrians.  These are the objects in scene that are active or pooled.  Iterate though these, getting the data you need. Be sure to check if they are active in scene
16) Create a data class similar to what we use for GroundTruth3D and send it over the bridge.  You may need some other minor data class changes but Simulator supports custom bridge message types.
17) Now when you run Simulator, you can add your sensor in the vehicle sensor parameters, and it will spawn at run-time.

Hopefully this helps.

vatna, 2020-03-03:
Thanks for all those instructions. I'll work my way through them!

closed 2020-03-03 by vatna

fabriquant, 2020-03-10:
Hello @EricBoiseLGSVL,

I am trying to follow your steps and I have a question regarding your suggested step 6, where I got stuck.

> 
>     6. Rename JSONGroundTruth3D class inside the namespace to JSONGroundTruthScene
> 

Where should I look for the JSONGroundTruth3D ? I could not find it in the repository.

Many thanks for your help

EricBoiseLGSVL, 2020-03-10:
@fabriquant No problem, glad to help.  6) is a data class found in the GroundTruth3DSensor class you duped.  You need to rename or it won't compile

fabriquant, 2020-03-10:
Thanks for you quick reply @EricBoiseLGSVL , this is awesome!
 
I looked for JSONGroundTruth3D in all files (even using grep) and I could not find it...

Could it exists in your fork of the repository and not on Github? 
Or should it be generated by other tools?

EricBoiseLGSVL, 2020-03-10:
@fabriquant 
[image]

fabriquant, 2020-03-10:
Thank you @EricBoiseLGSVL ,

I have attached a screenshot of 
simulator/Assets/Scripts/Sensors/GroundTruth3DSensor.cs under the **master** branch. 

Do you also see that the class JSONGroundTruth3D is not implemented on version f0a7aa8422034 (my git HEAD) ?

[image]

EricBoiseLGSVL, 2020-03-10:
@fabriquant Ah, sorry, this change must not be merged yet.  It's just a data class and needed to be changed to fix compile errors from duplicating the class.

fabriquant, 2020-03-10:
Thanks @EricBoiseLGSVL, in that case, I'll use your screenshot for the JSON dataclass.

fabriquant, 2020-03-11:
@EricBoiseLGSVL , thank you very much, I am now able to clear the compilation errors.

would you have an advice to effectively :
- perform Step 17. If I understand correctly, there is an option (tickbox in simulator environement/boolean in code) that must be set to add the sensor ?

- to debug my way through when adding references to the right objects ?

Many thanks

EricBoiseLGSVL, 2020-03-11:
@fabriquant no, 17 is adding sensor in parameters in WebUI like [so](https://www.lgsvlsimulator.com/docs/sensor-json-options/).

Use the existing ground truth options to set your new sensor.

fabriquant, 2020-03-19:
Thanks @EricBoiseLGSVL

fabriquant, 2020-04-01:
Hello @EricBoiseLGSVL 

I was able to generate logs from your guidance for a GroundTruthSensor (GT curve). 
This is a plot comparing the GroundTruthSensor to the IMU positions of the ego vehicle : 

[image]

Unfortunately, I observed a big time lag between the GroundTruthSensor log and data that I log to the disk directly from the IMU Sensor. 

This is an example of a drive log for the ego vehicle. The graph shows the change of x position of the car on the map as a function of the time (unix timestamp). The blue graph is the data I log from the IMU Sensor and the orange graph is the data logged from the GroundTruth sensor.
You can notice a time lag of roughly **8 seconds** between the two. In addition, the time lag is not constant, it jitters for about 10 seconds.
  
[image]

To produce this graph I followed your instructions to make a GroundTruth sensor that uses the WhileInRange callback to push updates to a filestream. The stream is flushed about every half a seconds. I also modified the ImuSensor.cs file (in the FixedUpdate()) to save the different ego vehicle positions. The IMU log is very robust in terms jitter (one entry every 10 ms). But the  GroundTruth sensor WhileInRange does seem to produce lagging data.

[image]

+ Would you expect a time lag to occur when GroundTruth WhileInRange callback ?
+ In your opinion, what would cause this lag ?
+ What would be a robust way to fix it ?
   -  For example, would logging directly from the NPC controllers PhysicsUpdate help ?

Many thanks in advance for your help

EricBoiseLGSVL, 2020-04-01:
@fabriquant The lag is from GT sensor writer being in Update while IMU sensor writer is in FixedUpdate.  I would move GT sensor to FixedUpdate but be aware that this will effect performance.  **Note** Move to FixedUpdate in sensor NOT in NPCController FixedUpdate.

fabriquant, 2020-04-02:
Hello @EricBoiseLGSVL,

This seems to be the fix : I only changed my timestamp calculation in the GroundTruthSensor (WhileInRange callback) from :

```
var unixEpoch = new DateTime(1970, 1, 1, 0, 0, 0, System.DateTimeKind.Utc);
double Timestamp = (DateTime.UtcNow - unixEpoch).TotalSeconds;
```
 
to:

`double Timestamp = SimulatorManager.Instance.CurrentTime;`

I am a C# and Unity beginner and I am still very surprised by the fact I got a jitter of 2 seconds on top of a lag of 10 seconds when calculating the timestamp from scratch.

Many thanks!

EricBoiseLGSVL, 2020-04-02:
@fabriquant Nice! Glad you were able to find a fix.  This makes sense.  You needed to increment Timestamp in fixedUpdate, which we are doing already in SimulatorManager, and initialize TimeStamp in Awake, not every frame or fixed frame.