lgsvl/simulator #664: How to restore default setting

atinfinity, 2020-03-02:
I tried to use LGSVL Simulator 2020.01 on Ubuntu 18.04.
But, I accidentally deleted the setting of Map and Vehicle from Web UI.
Is there a way to restore default setting?

EricBoiseLGSVL, 2020-03-02:
@atinfinity Navigate to the project persistent data path and just delete the data.db and backup.db.  Download the latest exe and run it.  It will re-download all settings and maps/vehicles.

This is my windows path
[image]

atinfinity, 2020-03-02:
@EricBoiseLGSVL Thank you for your reply. It's usefull information!

> I tried to use LGSVL Simulator 2020.01 on Ubuntu 18.04.

I found setting file in `${HOME}/.config/unity3d/LG Silicon Valley Lab`.
So, I removed this directory. As a result, I could restore default setting.

closed 2020-03-02 by atinfinity