lgsvl/simulator #666: Custom or Unity assets work in a Unity project but not in the simulator project

asgardus, 2020-03-03:
I am trying to create custom scenarios with externally created assets and also assets from the Unity store. The custom assets like vehicle body, traffic signs, trees etc are mostly in the .fbx format while some tree assets I imported from Unity Store are .spm format.

The external or store assets imported and work fine in a normal Unity project, but when imported into the simulator project they become pink. The shaders are assigned and the materials and textures are present, they are just pink in the simulator project but work well in a normal project.

I have tried extracting and regenerating textures and materials but without success. I am attaching pictures to help explain the situation better.

[image]
[image]
[image]
[image]

Any tips or help on this would be great, Thanks!

MikeLGSVL, 2020-03-03:
@asgardus The SPM file, is a Speedtree file. Change the shader type to HDRP/Lit and on the BroadLeaf tree object, change the Fade Mode to Cross Fade or None. Whichever better suites your project.

EricBoiseLGSVL, 2020-03-03:
@asgardus Simulator is an HDRP project.  You will need to convert any assets to HDRP materials.
Edit -> Render Pipeline -> Upgrade project materials to High Definition materials.
Most purchased assets with work fine once updated except custom shaders.  Those will need to be rewritten or use an equivalent HDRP shader and settings.

asgardus, 2020-03-04:
Thanks for the help!

closed 2020-03-04 by asgardus