lgsvl/simulator #667: GpsOdometrySensor publishes inaccurate orientation and velocity.

ozzdemir, 2020-03-04:
First of all, I am using autoware and lgsvl( revision 48650e7) in CubeTown.
I observed this problem when I compared orientations found by ndt_maching and odom published by the GpsOdometrySensor.
Does this happen to anyone else?
My solution procedure to this problem is as follows.

Btw, sensor config is as follows.(Configured with respect to change in position)
```
{
    "type": "GPS Odometry",
    "name": "GPS Odometry",
    "params": {
      "Frequency": 12.5,
      "Topic": "/vehicle/odom",
      "Frame": "gps",
      "IgnoreMapOrigin": true
    },
    "transform": {
      "x": 0.0,
      "y": 0.0,
      "z": 0.0,
      "pitch": 0,
      "yaw": 270,
      "roll": 0
    }
  }
```
The problem is due to unity being left-handed (and conversion is not implemented). I made it work with following changes.
https://github.com/lgsvl/simulator/blob/cb937deb8e633573f6c0cc76c9f451398b8b9eff/Assets/Scripts/Bridge/Ros/RosConversions.cs#L708-L711
to Following. It converts from left handed to right handed. From [here](https://gamedev.stackexchange.com/a/157954)
```
        static Quaternion Convert(UnityEngine.Quaternion q)
        { 
            return new Quaternion() { x = -q.z, y = q.x, z = -q.y, w = q.w };
        }
```
Also linear velocity was inaccurate (this one might be caused by coordinate frame of the sensor, if it is; it might need to be handled as well.).
Changed
https://github.com/lgsvl/simulator/blob/cb937deb8e633573f6c0cc76c9f451398b8b9eff/Assets/Scripts/Sensors/GpsOdometrySensor.cs#L185
to
```
ForwardSpeed = Vector3.Dot(RigidBody.velocity, transform.right),
```
Since conversion of imu was handled in the sensor module, I changed it as follows.
from
https://github.com/lgsvl/simulator/blob/cb937deb8e633573f6c0cc76c9f451398b8b9eff/Assets/Scripts/Sensors/ImuSensor.cs#L130-L153
to
```
var angularVelocity = RigidBody.angularVelocity;

            data = new ImuData()
            {
                Name = Name,
                Frame = Frame,
                Time = time,
                Sequence = Sequence,

                MeasurementSpan = Time.fixedDeltaTime,

                Position = transform.position,
                Orientation = transform.rotation,

                Acceleration = acceleration,
                LinearVelocity = velocity,
                AngularVelocity = angularVelocity,
            };
```

EricBoiseLGSVL, 2020-03-04:
@hadiTab has found this as well and is looking into a fix

EricBoiseLGSVL, 2020-04-17:
@ozzdemir we have a fix in the next release.

ozzdemir, 2020-04-29:
@EricBoiseLGSVL, by next release, did you mean release-2020.03?
In 2020.03, I saw changes on ImuSensor but did not see related changes on GpsOdometerSensor and also its' release date is quite close to the date of your message, that's why I am confused :sweat_smile:
Btw I didn't try yet but just reviewed the changes.

EricBoiseLGSVL, 2020-04-29:
@ozzdemir we made some changes to the conversions and I think it is in 2020.03, I'll check with @hadiTab

hadiTab, 2020-04-29:
Sorry, it is not in 2020.03. The orientation issue for IMU was only fixed in 2020.03. We are starting work on the GPS Odometery orientation issue now, so hopefully next release. Thanks.

EricBoiseLGSVL, 2020-04-29:
Thanks @hadiTab .  @ozzdemir We will be releasing 2020.04 soon with this fixed

deepaktalwardt, 2020-07-18:
Has this been fixed in the 2020.05 release?

EricBoiseLGSVL, 2020-07-21:
@deepaktalwardt I think this is in the next release.  Right @hadiTab ?

hadiTab, 2020-07-21:
Yup. Next release.

deepaktalwardt, 2020-07-21:
@hadiTab do you know when that will be? Currently I'm stuck with incorrect odometry that's messing up my algorithm.

closed 2020-10-06 by EricBoiseLGSVL