lgsvl/simulator #669: cannot open web browser localhost:8080 after launching LGSVL simulator

ck-leo, 2020-03-05:
I have encountered the problem that I cannot open browser when I click the button "open browser ..." in the LGSVL simulator window.
[image]
[image]
And in the website https://account.lgsvlsimulator.com, I have signed up my account. But I cannot help.
[image]
My error log is :
[Player.log](https://github.com/lgsvl/simulator/files/4290376/Player.log)

My platform is as following:
Windows10
i7 9750H @2.60GHz
16G DDR4
GTX 2060 6G
lgsvlsimulator-windows64-2020.01
@EricBoiseLGSVL 
What do mean when you say "please post what you are on Linux/Windows/ReleaseBuild/Editor/CreatedBuild"? I'm sorry I can't follow.
I havenot installed the related components of unity, it that the problem? Or is there some other suggestings?
Thanks.

ntutangyun, 2020-03-05:
@ck-leo Ali-ren ?  you need to build the frontend before opening it in browser.

try:
```bash
#make sure you have node and npm installed
cd simulator_project_root/WebUI/
npm install
npm run pack
```

ck-leo, 2020-03-05:
@ntutangyun because I use ali source, then I registered an ali-email.
I have installed node and npm, and then run the command: 
cd simulator_project_root/WebUI/
npm install
npm run pack
But I still could not open the browser .

EricBoiseLGSVL, 2020-03-05:
@ck-leo this is an issue with mono.  See [here](https://github.com/mono/mono/issues/7117)

ck-leo, 2020-03-06:
@EricBoiseLGSVL Yeah, that works , thank you so much.
And @ntutangyun thank you all the same.

EricBoiseLGSVL, 2020-03-06:
@ck-leo You are welcome.

closed 2020-03-06 by EricBoiseLGSVL

hiber-wang, 2022-11-29:
@ck-leo 
Hello, I have encountered the same problem, how did you solve it?