lgsvl/simulator #671: How can I create my own map?

ctgloglu, 2020-03-05:
How do I create my own map in unity

ntutangyun, 2020-03-05:
how about starting with loading one of the existing maps and messing things around ?

ctgloglu, 2020-03-05:
First of all thank you for your answer.As far as I can see on the internet, maps are loaded from the simulator menu.I don't have that menu either.How do I install the simulator menu unity.

ntutangyun, 2020-03-05:
@ctgloglu welcome. 
the maps are usually loaded from AssetBundles, which are available at [lgsvl Content website](https://content.lgsvlsimulator.com/maps/).  You could download existing maps and vehicles there.

Once you've downloaded the maps, you could 
1. start simulator in Unity Engine, 
2. add new map entries in simulator webUI, with the address to the downloaded map assetbundle files.
3. run the simulator with one of the maps. 
4. in Unity Editor, once the map is loaded, everything that the map contains will be available in the `Hierarchy` tab.

ctgloglu, 2020-03-05:
[image]

There is no simulator menu. How can I install

ctgloglu, 2020-03-05:
https://github.com/lgsvl/simulator
I get an error when integrating the files in unity
[image]

luqiang21, 2020-03-05:
@ctgloglu Did you install git lfs?

ctgloglu, 2020-03-06:
I tried to install according to the instructions written here: https://www.lgsvlsimulator.com/docs/build-instructions/

* git lfs install  output: 
Git LFS initialized.

*Window ex. C:\Users\XXX\Documents\Simulator\WebUI 

* npm install output:

audited 885770 packages in 12.523s
found 15999 vulnerabilities (3 moderate, 15995 high, 1 critical)
  run `npm audit fix` to fix them, or `npm audit` for details

ctgloglu, 2020-03-06:
I solved problem. https://www.lgsvlsimulator.com/docs/build-instructions/ I was trying to do my step manually in the 5th instruction here.Fixed the problem when doing it with cmd screen.Thank you to friends who are interested in my problem.

closed 2020-03-06 by ctgloglu