lgsvl/simulator #672: error response from daemon when bash ./docker/scripts/dev_into.sh

ck-leo, 2020-03-05:
I followed the instruction in the website https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/, and got an error when  running bash ./docker/scripts/dev_into.sh. Details are as following :
[image]
And my platform is :
Ubuntu 16.04
i3 8100 @3.60GHzX4
16G DDR4
Intel® HD Graphics (Coffeelake 3x8 GT2) (Yes , I donot have a Nvidia graphics)
I want to know if I there are some methods to run apollo 5.0-lgsvl without Nvidia graphics? Thanks.

EricBoiseLGSVL, 2020-03-06:
@ck-leo I don't think non Nvidia is supported. @hadiTab ?

lemketron, 2020-03-06:
Apollo may or may not require an Nvidia GPU (e.g. if you don't need perception, for example) but our setup instructions assume nvidia-docker. Could you try installing nvidia-docker to see if that gets you past this particular error? If so, you'll then need to avoid enabling Apollo perception (which does require GPU).

lemketron, 2020-03-07:
We have confirmed that with Apollo 5.0, docker has to have nvidia-docker. So it is not possible to run with Intel/AMD GPU. A CPU-only version Apollo is not under maintenance any more.

ck-leo, 2020-03-08:
@lemketron Thank you very much, then I should get an Nvidia GPU.

ck-leo, 2020-03-08:
@lemketron @EricBoiseLGSVL  Thanks very much, then I should get an Nvidia GPU.

closed 2020-04-16 by EricBoiseLGSVL