lgsvl/simulator #673: Custom sensor: Types are not converted

joshwapohlmann, 2020-03-06:
In 2020.01 I created a custom sensor and used the ComfortSensor as starting point.
I implemented my own data type and everything. The problem is, that the data is not serialized properly.
Debugging showed, that in `RosBridge.cs` l. 312 writers are created with `writer = new Writer<T, object>`. This causes the serialize function (same file) in l. 743 to fail getting the fields of said data type, since it gets a `System.Object` instead of my actual data type.
Please fix this.

joshwapohlmann, 2020-03-06:
Lines in Question
https://github.com/lgsvl/simulator/blob/cb937deb8e633573f6c0cc76c9f451398b8b9eff/Assets/Scripts/Bridge/Ros/RosBridge.cs#L312
https://github.com/lgsvl/simulator/blob/cb937deb8e633573f6c0cc76c9f451398b8b9eff/Assets/Scripts/Bridge/Ros/RosBridge.cs#L743

EricBoiseLGSVL, 2020-03-06:
@mark-gerow-lge What is @joshwapohlmann missing in the custom sensor?

joshwapohlmann, 2020-03-24:
@mark-gerow-lge Hello, any chance you could shed some light into this? We really need custom sensors/actors and their custom message types!

joshwapohlmann, 2020-04-06:
We fixed it ourself now. My guess was correct. Creating a new generic type at runtime does the trick.

EricBoiseLGSVL, 2020-04-06:
@joshwapohlmann We have a fix for this as well but we are still working on a new public release.  Glad you found a solution

dr563105, 2020-05-11:
@joshwapohlmann 
Hello!
I'm doing something similar to what you did -- I have extended [Lanefollowing](https://github.com/lgsvl/lanefollowing) tutorial and its [sensor plugin](https://github.com/lgsvl/LaneFollowingSensor) to my needs. This led to using custom message types to transfer data. I'm running into a few problems building the simulator in Unity. I suspect it is do with the custom messages declared inside the LanefollowingData.cs(see the gist link). I'm sending from the simulator through the ROS2 bridge and to the [python script](https://github.com/lgsvl/lanefollowing/blob/master/ros2_ws/src/lane_following/collect.py). 
If it is not much trouble, could you, please, have a look at my files and point me out how to solve those issues? 
https://gist.github.com/dr563105/3bd38e12febf0b87a2eda8f345b2fde9

Thanks.

EricBoiseLGSVL, 2020-05-12:
@dr563105 we have a fix and should be releasing it soon

joshwapohlmann, 2020-05-14:
@dr563105 We fiddled in some ugly casting using introspection in `RosBridge.cs`. Without this, we couldn't make it work. So, as soon as you implement custom data types and don't fix LGSVLs broken casting system for those, I don't see a way for you.
However, you should be able to BUILD the simulator regardless this issue. The problem only reveals itself by not sending any actual data.

EricBoiseLGSVL, 2020-05-14:
@dr563105 @joshwapohlmann We will have a new release this month with a fix.  I am including the diff for the changes that you both can use in the meantime.  Be aware that this hasn't been through our release testing but I hope this helps
[CustomTypeFix.zip](https://github.com/lgsvl/simulator/files/4629757/CustomTypeFix.zip)

dr563105, 2020-05-14:
Thanks @EricBoiseLGSVL. When @daviduhm adapted the Lanefollowing tutorial for the latest simulator, he advised me to these changes. However, I was facing issues when I extended that custom sensor plugin to my needs. Hence I contacted @joshwapohlmann for help. 

Thank you @joshwapohlmann for your reply.

EricBoiseLGSVL, 2020-05-14:
@dr563105 Good.  Please let us know what is causing issues with the custom sensors or new features needed with the system, we would like to make this feature better and any insight to improve it will help other users.

closed 2020-10-06 by EricBoiseLGSVL