lgsvl/simulator #674: npm install error

ctgloglu, 2020-03-06:
I get an error when I install npm.

[image]

EricBoiseLGSVL, 2020-03-06:
@ctgloglu What platform are you on?

ctgloglu, 2020-03-06:
Windows 10

EricBoiseLGSVL, 2020-03-06:
What version of Node.js do you have?
Can you run the executable release build?

ctgloglu, 2020-03-06:
node.js version v12.16.1

EricBoiseLGSVL, 2020-03-06:
I think those are just warnings.  Can you run> npm run pack
What output do you get?

ctgloglu, 2020-03-07:
> simulator-web-app@1.0.0 pack C:\Users\Can\Desktop\deneme-github\simulator\WebUI
> enact pack

Creating a development build...
Browserslist: caniuse-lite is outdated. Please run next command `npm update`
        5.21 MB         dist\main.js
        35.92 KB        dist\main.css

Compiled with warnings:

./src/components/SimulationManager/FormTraffic.js
  Line 18:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 27:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 28:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 29:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps

./src/components/SimulationManager/FormGeneral.js
  Line 23:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 24:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 25:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 26:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 30:  Assignments to the 'unmounted' variable from inside React Hook useEffect will be lost after each render. To preserve the value over time, store it in a useRef Hook and keep the mutable value in the '.current' property. Otherwise, you can move this variable directly inside useEffect  react-hooks/exhaustive-deps

./src/components/SimulationManager/FormMapVehicles.js
  Line 26:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 27:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 28:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 46:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 126:  'vehicleList' is already declared in the upper scope                                                                      no-shadow

./src/components/SimulationManager/FormWeather.js
  Line 27:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 32:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 41:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 50:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 59:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps

./src/components/MapManager/MapManager.js
  Line 33:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 34:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 61:   Assignments to the 'unmounted' variable from inside React Hook useEffect will be lost after each render. To preserve the value over time, store it in a useRef Hook and keep the mutable value in the '.current' property. Otherwise, you can move this variable directly inside useEffect  react-hooks/exhaustive-deps
[... cut ...]
  Line 62:  Expected '===' and instead saw '=='  eqeqeq

./src/components/SimulationManager/SimulationManager.js
  Line 63:  Assignments to the 'unmounted' variable from inside React Hook useEffect will be lost after each render. To preserve the value over time, store it in a useRef Hook and keep the mutable value in the '.current' property. Otherwise, you can move this variable directly inside useEffect  react-hooks/exhaustive-deps

NOTICE: This build contains debugging functionality and may run slower than in production mode.

closed 2020-03-07 by ctgloglu

Julyers, 2020-03-24:
Same, how did you solve it?

EricBoiseLGSVL, 2020-03-24:
Did you run npm update?

Julyers, 2020-03-24:
> Did you run npm update?

Yes, I run. But when I build WebUI in unity, it still show erros as follow.
**Failed to build WebUI. Please make sure npm is installed. Check README.md file
UnityEngine.Debug:LogError(Object)
Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:40)**

Julyers, 2020-03-24:
Who can help me? 
[image]

ntutangyun, 2020-03-24:
@Julyers  Please refere to [a possible solution here](https://github.com/lgsvl/simulator/issues/388#issuecomment-536516345)

Julyers, 2020-03-25:
> Creating a development build...
> Browserslist: caniuse-lite is outdated. Please run next command `npm update`
> 5.21 MB dist\main.js
> 35.92 KB dist\main.css
>
> Compiled with warnings: [...]
Hello, how did you solve this problem in the end? My mistake is exactly the same as yours. I've been stuck in this place for a long time. I hope you can tell me how you solved it, thank you!

Julyers, 2020-03-25:
> @Julyers Please refere to [a possible solution here](https://github.com/lgsvl/simulator/issues/388#issuecomment-536516345)

I tried it but it didn't work. And same problem still exists. Thank you！

martins-mozeiko, 2020-03-25:
Try running npm manually in command line inside WebUI folder: `npm run pack-p` This will show more detailed error message.