lgsvl/simulator #676: Map error

ctgloglu, 2020-03-07:
I can't see objects, it gives an error

[image]

- Shader error in 'Shader Graphs/EnvironmentSimulation': 'NewGradient': no matching 0 parameter function at line 4839 (on d3d11)

Compiling Fragment program with INSTANCING_ON
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

- Shader error in 'Shader Graphs/EnvironmentSimulation': syntax error: unexpected integer constant at line 4839 (on d3d11)

Compiling Fragment program with INSTANCING_ON
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

- Shader error in 'Shader Graphs/EnvironmentSimulation': 'NewGradient': no matching 0 parameter function at line 7359 (on d3d11)

Compiling Vertex program with INSTANCING_ON DECALS_3RT LIGHT_LAYERS
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

- Shader error in 'Shader Graphs/EnvironmentSimulation': syntax error: unexpected integer constant at line 7359 (on d3d11)

Compiling Vertex program with INSTANCING_ON DECALS_3RT LIGHT_LAYERS
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

shalinmehtalgsvl, 2020-03-09:
Hi @ctgloglu ,

It looks like you are missing the materials or textures. In the Project Panel, in SingleLaneRoad>Models>SingleLaneRoad>Materials, there should be a SingleLaneRoadDiffuse.png. Can you see the image?

If not, did you have git lfs installed before you cloned the repo? You'll need to do `git lfs install` and then `git pull` to get the missing files.

JockTokk, 2020-03-20:
Hello, I have the same problem with SanFrancisco
[image]

closed 2020-04-08 by EricBoiseLGSVL