lgsvl/simulator #677: LiDAR data delayed against vehicle

atinfinity, 2020-03-08:
### System information (version)
- LGSVL Simulator => lgsvlsimulator-linux64-2020.01
- Operating System / Platform => Ubuntu 18.04 64bit
- NVIDIA Driver => nvidia-driver-440
- Vulkan loader library => libvulkan1(1.1.70+dfsg1-1ubuntu0.18.04.1)
- GPU => NVIDIA GeForce RTX 2080 Ti

### Detailed description
I visualized LiDAR data on LGSVL Simulator.
But, LiDAR data delayed against vehicle.

[image]

### Steps to reproduce
I launched LGSVL Simulator.

```
$ ./lgsvlsimulator-linux64-2020.01/simulator
```

I selected the following setting.

- Screen resolution: 1024 x 768
- Graphics Quality: Ultra

1. I selected `BorregasAve, no bridge, data collection` via Web UI. And, I changed to WideFlatMap and played simulator.
1. I enabled the visualization of LiDAR.
1. I operated vehicle by keyboard.

rongguodong, 2020-03-09:
Can you explain more on your step 1? Do you mean this only happens if you run BorregasAve first, and then run WideFlatMap? How do you "changed to WideFlatMap"?

atinfinity, 2020-03-09:
@rongguodong Sorry for not explaining enough. I attached screenshot.

> I selected Simulator(BorregasAve, no bridge, data collection) via Web UI.

[image]

> And, I changed to WideFlatMap.

[image]

rongguodong, 2020-03-09:
I tried the latest build on my Linux machine and cannot reproduce this issue. See my screenshot as below:
[image]

rongguodong, 2020-03-09:
What I did is:
* Download the latest build: https://github.com/lgsvl/simulator/releases/download/2020.01/lgsvlsimulator-linux64-2020.01.zip
* Unzip and run it
* Click "Open Browser" button to open the WebUI. (All assets will be automatically downloaded)
* Go to "Simulation" page and select the first simulation. Change its map to "WildFlatMap", and run it.

If you did anything different, please let me know.

atinfinity, 2020-03-09:
I turned up the vehicle speed by the operation of keyboard.

rongguodong, 2020-03-10:
If the speed is too fast, maybe the Lidar points are not generated fast enough to catch the ego car. As long as the ego car slow down or stops, the Lidar points should be correct.

atinfinity, 2020-03-16:
@rongguodong Thank you for your reply. Is there a workaround besides slowing down the car?

EricBoiseLGSVL, 2020-04-16:
@atinfinity Most AD stacks we support do not support EGO vehicles at high speed.  We are working on RTX support and UDP LiDAR sensors to improve performance but this will be in a future release

atinfinity, 2020-04-17:
@EricBoiseLGSVL Thank you for your reply. I understood about future plan.

closed 2020-04-17 by EricBoiseLGSVL

ZeKubiki, 2020-12-10:
@EricBoiseLGSVL just curious how it's going with the RTX and UDP Lidar updates. Very much looking forward to it. Do you have an expected release date?

EricBoiseLGSVL, 2020-12-11:
@ZeKubiki we have a UDP Velodyne Lidar solution now but it is not open source at the moment.  We are still working on issues with RTX and Linux.