lgsvl/simulator #678: Can't run Rviz with autoware

Amina88, 2020-03-10:
$rosrun rviz rviz
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-sneibus'
[ INFO] [1583843544.906934800]: rviz version 1.13.7
[ INFO] [1583843544.907294400]: compiled against Qt version 5.9.5
[ INFO] [1583843544.907930200]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1583843544.914675100]: Forcing OpenGl version 0.
Segmentation fault (core dumped)

EricBoiseLGSVL, 2020-03-16:
@Amina88 This looks like an issue with your gpu driver and Rviz.  Any ideas @luqiang21 ?

luqiang21, 2020-03-18:
@Amina88 Sorry I don't have experience on this, maybe you can refer https://github.com/ros-visualization/rviz/issues/975#issuecomment-419907750

EricBoiseLGSVL, 2020-05-08:
@Amina88 Are you still having this issue?  Did @luqiang21 post help?

closed 2020-07-31 by lemketron