lgsvl/simulator #680: Request for Clarification - LGSVL Simulator

guruvishnuvardan, 2020-03-11:
Hi,

I am part of a startup from India and we are working on **Semi Autonomous Car Simulation for Indian Roads.**

We have created our own AD algorithm(for simulator) and we have started testing using the following Code - https://github.com/Kyushik/DRL (it's an AD simulation using Unity and DRL), but we realized, we need to run our simulations in a more real time simulator 

Can you please clarify me the following:

1) Can we use, Our own **Semi-AD** algorithm(that suits Indian Driving Conditions), instead of Apollo or Autoware?, we are only using radar, to detect obstacles and we do not mark lanes or use maps, Is it possible to Simulate in LGSVL without lanes and maps?, If so, Can you please let me know the process to do that.

2)  We are planning to use NVIDIA 1080 TI(11 GB VRAM),Intel i7 processor (7th Gen), 32 GB RAM,Windows 10, Can you please let me know, if I can use a standalone system for LGSVL Simulator.

3) Do we need to use Unity or Is LGSVL itself support a completely customized simulation of Semi-AD

Thanks
Guru

EricBoiseLGSVL, 2020-03-11:
@guruvishnuvardan @zelenkovsky Glad you are looking to use LGSVL Simulator.
1) We created LGSVL Simulator to use any AD stack.  You would need to make improvements to the radar sensor and just use our default map object that is created on new scene.  This has no HDMap data, just the needed game objects to compile.
2) This spec is sufficient, but be aware that the size and scope of the map effects performance.
3) Unity is required if you plan on changing source code or make assets other than the ones provided.

guruvishnuvardan, 2020-03-12:
Thanks @EricBoiseLGSVL :)

closed 2020-03-19 by EricBoiseLGSVL