lgsvl/simulator #681: Can I add a vehicle with a local path?

mamemo888, 2020-03-11:
Hello,
I tried to add the vehicle AssetBundle modified by UnityEditer in WebUI.
However, despite entering the local path to the binary, the message "Failed to add vehicle: You must specify a non-empty name" is displayed.

[image]

The documentation states:
"When adding a vehicle, the link to the appropriate assetbundle can be entered as the URL or the assetbundle can be downloaded manually and the local path can be entered."

I have just started using this simulator.
Am I misunderstanding something?
Thank you.

EricBoiseLGSVL, 2020-03-11:
@mamemo888 yes you can add a vehicle from a local path but it looks like you didn't build bundle with our editor tool or you renamed.  Please see [this](https://www.lgsvlsimulator.com/docs/assets/)

mamemo888, 2020-03-12:
@EricBoiseLGSVL 
Thank you for your reply.
However, I think the bundle was built. A zip file is created in the AssetBundle folder. The asset and folder names also match.

By the way, I have two new questions.
Is the location to extract the zip of the asset bundle specified?
Is the local path I need to enter in “Add new” the path to the unpacked binaries? Or something else?

EricBoiseLGSVL, 2020-03-12:
@mamemo888 
1) Asset bundle file name should be vehicle_NewVehicle if successful.
2) The folder that contains the assets that you build in editor should be Assets/External/Vehicles/NewVehicle
3) The name of the vehicle prefab should be NewVehicle.

Does your project match this naming convention before you build?
This is required because Unity needs filenames when dealing with asset bundle loading.

mamemo888, 2020-03-13:
@EricBoiseLGSVL 
The build may not have been completed correctly, so I tried again and confirmed your point.
Look at the image below for point 1.
[image]
The file name of AseetBundle is “vehicle_new_vehicle”.

See the image below for the contents of this zip file. Are these correct?
[image]

For points 2 and 3, see this image.
[image]
The asset folder is “Assets / External / Vehicles / new_vehicle”.
The name of the prefab is “new_vehicle.prefab”.

There were no errors when building with UnityEditer. Are these not built correctly?

EricBoiseLGSVL, 2020-03-13:
@mamemo888 That looks exactly right.  Now when you link the bundle in the webUI, does it give you an error?

mamemo888, 2020-03-16:
@EricBoiseLGSVL 
I entered the file name of the bundle, but unfortunately the error did not go away.
[image]

mamemo888, 2020-03-16:
@EricBoiseLGSVL 
By the way, I also tried the asset bundle downloaded from the official website.
https://content.lgsvlsimulator.com/vehicles/lincolnmkz2017/

However, this also showed the same error.
[image]

I feel like I'm making a fundamental mistake.

mamemo888, 2020-03-16:
@EricBoiseLGSVL 
After all I did not know the underlying mechanism.
I learned from another person that I had to build an http server. I did this and solved the problem.

I am very grateful for your response so far.

closed 2020-03-16 by mamemo888

martins-mozeiko, 2020-03-16:
The problem is that you are unzipping and providing `XXX_vehilce_main_linux` file to simulator. You should provide path to whole "zip archive" file.