lgsvl/simulator #682: Integrating UnityCloud and AWS

RanjaniN-10, 2020-03-12:
Hi LG team,
I am  working with LGSVL Simulator.
I would like to modify this simulation, but it requires high-end hardware. So I prefer to use UnityCloud along with AWS.

Is it possible to integrate UnityCloud and  AWS for LGSVL Simulator training? If so can you help me with that.

EricBoiseLGSVL, 2020-03-16:
@RanjaniN-10 This is not supported at this time.  We are currently developing these features.

guruvishnuvardan, 2020-03-23:
Hi, 
Can I train LGSVL Simulator using Unity and Standalone windows GPU system?

EricBoiseLGSVL, 2020-03-23:
@guruvishnuvardan @daviduhm do we have documentation for training with simulator?

guruvishnuvardan, 2020-03-23:
@EricBoiseLGSVL - No. I don't have. I am new to LGSVL Simulator. Can you please share me the links to get the documentation for training with LGSVL Simulator

lemketron, 2020-03-23:
> @EricBoiseLGSVL - I am new to LGSVL Simulator. Can you please share me the links to get the documentation for training with LGSVL Simulator

If you're looking for general documentation on setting up and using LGSVL Simulator, you can find it here:
https://www.lgsvlsimulator.com/docs/
https://www.lgsvlsimulator.com/docs/getting-started/

guruvishnuvardan, 2020-03-25:
@lemketron - Thanks a ton :)

One other question - Can I use Microsoft Azure Virtual Machine and Install LGSVL Simulator for Training?

Since, We were trying to Use Unity Cloud Beta - AWS - LGSVL for training our own simulation, but I heard, We can't use AWS to train LGSVL Simulator.

Can you please suggest us a cloud platform(Virtual Machine with GPU) to install Unity -LGSVL and start training.

Thanks
Guru

EricBoiseLGSVL, 2020-03-26:
@guruvishnuvardan Cloud is not supported at this time.  We are working on this feature.

martins-mozeiko, 2020-03-26:
If you can access Remote Desktop on machine with GPU the simulator will run just fine. Exactly same as on local machine. See following issue where running in Azure was discussed: #236

closed 2020-05-08 by EricBoiseLGSVL