lgsvl/simulator #683: build vehicle AsserBundle failed

Allenhe123, 2020-03-13:
when doing the  build step, i got an error:
FileNotFoundException: Could not find file "/home/abc/simulator/build/AssetBundles/7e351f8f-d2f1-4221-a1ff-b6a3fe871490_environment_main_windows"

I have cloned a map and a vehicle from your repository into simulator/Assets/External/Environments/SanFrancisco  and simulator/Assets/External/Vehicles/Jaguar2015XE

My build directory is simulator/build/.

A map-bundle  simulator/build/AssetBundles/environment_SanFrancisco  was generated successfully. But there is no vehicle generated.

What's wrong?

EricBoiseLGSVL, 2020-03-13:
@Allenhe123 sometimes building multiple bundles at once can cause issues, especially San Francisco because it has so many assets.  Can you try to just build the vehicle?  Did you make any changes to vehicle prefab or repo?

Allenhe123, 2020-03-24:
> @Allenhe123 sometimes building multiple bundles at once can cause issues, especially San Francisco because it has so many assets. Can you try to just build the vehicle? Did you make any changes to vehicle prefab or repo?

Appreciate your valuable reply, Eric. I did not do any change to vehicle prefab or repo.
I will try it.

EricBoiseLGSVL, 2020-04-08:
@Allenhe123 did building one bundle at a time help?  We are refactoring bundle building to have better performance and it should release in the next few releases

closed 2020-07-31 by lemketron