lgsvl/simulator #684: Replacing own vehicle in simulator

asgardus, 2020-03-13:
I am using simulator release 2020.01. I went through the [documentation](https://www.lgsvlsimulator.com/docs/add-new-ego-vehicle/) and issues [1](https://github.com/lgsvl/simulator/issues/441) and [2](https://github.com/lgsvl/simulator/issues/401) about adding a new ego vehicle and had a query-

The documentation and issues seem to address the change of geometry and visual elements of a vehicle. What would be the process if one wants to replace a vehicle with custom scripts for modelling behavior and dynamics, still using wheel colliders? Are the scripts associated with the root vehicle object (Vehicle Dynamics, Vehicle Controller, DistributedObject etc..) unreplaceable or is there an API to integrate custom vehicle scripts with the simulator?

Thanks and Regards.

EricBoiseLGSVL, 2020-03-13:
@asgardus We are working on a release soon that lets users create their own C# dynamics class or FMI supported class.  This should be released soon.  In the meantime, yes, all those scripts are necessary to run the simulator.  You will need to edit these classes.

asgardus, 2020-03-16:
Thanks for responding, Is it possible to provide an estimate for when to expect this functionality?

EricBoiseLGSVL, 2020-04-01:
@asgardus We plan on a public release this week or next week with this feature

closed 2020-04-01 by asgardus